pbatard/rufus #2218: suggestion needed

titet11, 2023-04-06:
hello friend, will it be possible to add an option to disable the paging file and hibernation? This would be useful for being able to install Windows on flash drives that are low on space and slow to read.﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

pbatard, 2023-04-07:
This can easily be disabled after Windows is installed and it doesn't hinder installation, so it's outside of the scope of what I am planning to do in Rufus.

Rufus will **never** be designed to tweak Windows exactly how you like it. It's purpose is to address installation woes, and this is not one of them.

&rarr; Rejected

closed 2023-04-07 by pbatard

titet11, 2023-04-07:
@pbatard 
I will argue because I do not agree with your opinion:
I have a USB 2.0 and I don't have the budget to buy another one.
when trying to configure the system it is impossible to configure it because each window takes a few minutes to load, when it should load in seconds.
I know this is not so much a problem with USB 2.0, but with the paging file, since I was using a 14gb USB 2.0 before, and due to low capacity, the system automatically disabled the paging file.
However, on that occasion when I installed the OS, I was able to run the OS very fast.
I am now installing windows on a 32gb usb 2.0 and due to capacity the paging file is enabled by default which prevents me from configuring the system because it gets slow.
so please add this option in rufus

pbatard, 2023-04-07:
> I know this is not so much a problem with USB 2.0,

It is. If you are trying to run Windows To Go on a USB 2.0 drive, you are 100% on your own because you are asking for trouble,

See also [this](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Why_cant_you_just_add_the_feature_I_want).

pbatard, 2023-04-07:
And I am saying that I am not going to waste my limited time addressing issue that I expect only you will have because you **chose** to run Windows on a less than subpar environment. For what is worth, _Windows To Go_ is no longer supported by Microsoft (most likely because they got tired of trying to satisfy people who insist on trying to run it on ill-suited hardware, such as USB 2.0 drives or drives that have slow random write I/O speed) and, while I am providing Windows To Go in Rufus, I am only doing so as a courtesy _"AS IS"_ (meaning that, just like Microsoft, I am not going to go out of my way trying to support people who insist on trying to run it on ill-suited hardware).

If you are that hell bent on customizing Windows To Go just the way you like it so that it runs on subpar hardware, I will encourage you to create your drive by following the official Microsoft procedure (which is what Rufus uses) at https://learn.microsoft.com/en-us/previous-versions/windows/it-pro/windows-8.1-and-8/jj721578(v=ws.11) and then apply whatever customization you want.

titet11, 2023-04-07:
@pbatard
First you have to respect that I don't have money to buy a high quality memory.
Second, I think you're wrong because as I told you before, if it was possible to run Windows on a USB 2.0:
 so you have more detailed information the system i ran on a 14gb usb 2.0 was windows 11 and as i said before while running windows 11 on a 14gb usb 2.0 it still allowed me to open videos and websites that could be useful to be able to use:
However, now that I put it on a 32gb usb based on the same hardware model, the pagefile.sys and iberfile.sys files remain active from the install, making the system so unusable, to the point where that it becomes impossible to configure the initial settings such as language, country, networks, among others.

Third, you should be aware that USB 2.0 is still widely sold today.

JonnyTech, 2023-04-08:
@titet11 consider Linux...

titet11, 2023-04-08:
@jonnytech  What I ask is not too much to ask.
2. Linux doesn't have the Intel HD Graphics driver, and also on Linux you have to issue commands and I don't like messing around with things I can do much easier on Windows.

titet11, 2023-04-08:
@pbatard  Don't be mean and put those 2 options that I asked for. You know well it's not too much to ask.

pbatard, 2023-04-08:
>  What I ask is not too much to ask.

Are you kidding me?

It'd be more profitable for me to buy and ship you a new USB 3.0 computer and flash drive than invest the time you're asking me to invest on that crusade of using that USB 2.0 drive of yours in conditions that will make the Windows experience supbar, no matter how much you want to pretend that _"it can run fine occasionally!"_

Please be very aware of this: It is not because I provide software for free that my __time__ is also free, and I'm afraid that, between losing money doing the thing you request (which is **not** a 5 minutes job as opposed to what you are thinking) or losing money fixing bugs and implementing features that will be beneficial to a lot more people, the choice is exceedingly clear.

I am not spending my days looking idly at the window and wondering what I could do next. As far as my private software development is concerned (which means that I am not talking about contractual work here) I **always** have a list long like the arm, of tasks that can keep me busy for the next 5 years without taking even a single requests or suggestions from users or finding new bugs/limitations to work around. So any time someone asks for something new to be added to that list, prioritization has to occur, and in this case, that prioritization, which is based in part on the amount of people that are likely to benefit from what's being requested, means that your request is never going to make it on my list.

Then again, and if you are unhappy that the developer of an **OPEN SOURCE** project is not adding the feature that you want, feel free to invest your **own** time cloning that project and adding that feature yourself. After all from where I stand, and I am making that statement with the knowledge that you may know nothing about programming and software development in the first place, it shouldn't be too much to ask (because there are loads of resources to help you out) and by all means, because Rufus is an Open Source application, if you want a new feature, you can always create your own version to add it (but of course this requires using your own free time, rather than asking a total stranger to use theirs for you).

titet11, 2023-04-12:
@pbatard  
there is something you are overlooking:
If I was able to run Windows 11 on a USB 2.0 drive, it's because my computer's ports are 3.1 and that almost triples the read/write speed.
However, as I said before, on a 32gb flash drive, it becomes impossible to set Windows initial options because of pagefile.sys and iberfile.sys.

My goal is not for you to work on Windows TO-GO compatibility, I'm just asking you to add the ability to disable pagefile.sys and iberfile.sys.
please help me with that sir.

pbatard, 2023-04-12:
> on a USB 2.0 drive, it's because my computer's ports are 3.1 and that almost triples the read/write speed.

_"I have a bicycle, but I ride it on the motorway, and that almost makes me reach the motorway speed limit when pedalling"_

That's the same level of technical absurdity as your claim above. Considering this, I'm going to stop giving you any credit and lock this thread altogether, especially as I already explained at length how your request is unneeded, since that's a one of operation that you can accomplish on your own even if you have to wait **a few minutes** (gasp!) to be able to do so and a fairly selfish request considering that I'm pretty confident you're the only person who wants that.