pbatard/rufus #2222: Rufus no longer lists older windows ISO's for download

ajawamnet, 2023-04-12:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

JonnyTech, 2023-04-12:
Windows 7 only missing? https://github.com/pbatard/Fido/issues/64

ajawamnet, 2023-04-12:
Yep that's it.  Thanks for the response...

On 4/12/2023 10:57 AM, JonnyTech wrote:
> Windows 7 only missing? https://github.com/pbatard/Fido/issues/64
>

-- 
WAM
http://www.ajawamnet.com
http://www.ajawamnet.com/ajawamnet/marketturd.htm
------------

More information on my lab and shop:
http://www.ajawamnet.com
http://www.ajawamnet.com/ajawam3/swarf/swarf.htm
------------------------

“In the information society, nobody thinks. We expected to banish paper, but we actually banished thought.”
~~ Michael Crichton, Jurassic Park

“You never change things by fighting the existing reality.
To change something, build a new model that makes the existing model obsolete.”
~~R. Buckminster Fuller (1895-1983)

"Success is going from failure to failure without a loss of enthusiasm."
~~ Sir Winston Leonard Spencer-Churchill (1874-1965)

"If the wrong man uses the right means, the right means work in the wrong way."
~~ Ancient Chinese Saying

"I've failed at so many things so many times I make Abraham Lincoln look like Bill Gates"
~~ Wayne Mitzen (1959-  )

"What makes an expert isn't so much what they know,
It's that they've done similar things so many times wrong
They know what not to do"
~~ Wayne Mitzen (1959-  )

"You just keep pushing. You just keep pushing. I made every mistake that could be made. But I just kept pushing."
~~ Rene Descartes 1596-1650

"I hate the way people just go from ignorance to despair so quickly; it's the lazy man's option"
~~Professor Tim Flannery (1956-  )

"When you do things right, people won't be sure you've done anything at all."
~~ God, in Futurama episode "Godfellas"

"Two things are infinite: the universe and human stupidity;
and I'm not even sure about the universe."
~~ Albert Einstein (1879-1955)

"Some scientists claim that hydrogen, because it is so plentiful, is the basic building block of the universe.
I dispute that. I say there is more stupidity than hydrogen, and that is the basic building block of the universe."
~~ Frank Zappa (1940-1993)

closed 2023-04-12 by pbatard