pbatard/rufus #2226: Rufus installed Windows 11 can not be upgraded with feature update on unsupported devices. Why this tool is allowing customization where people would stuck with receiving no updates?

sir-farts-alot, 2023-04-16:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

pbatard, 2023-04-16:
You should be able to install feature updates if you create a drive with Rufus and run the `setup.exe` from here.

And your "people would stuck with receiving no updates" is hyperbolic. People will receive Windows updates. And they will continue to receive updates after a feature release has been issued. It's just the **upgrade** from one feature release to the next that requires a bit more involved handling, but it is by no means is it impossible if you try a to use the same utility you used to install your Windows platform or look long enough on the internet.

Rufus expects that you have some knowledge of what you're doing or that you will be able to use trial and error or perform a search to accomplish what you want. It is not a utility that is designed to hold your hand.

closed 2023-04-16 by pbatard

pbatard, 2023-04-17:
> What I'm trying to say is that not receiveing any updates is a critical issue which all of those users gonna face anyway.

And what I'm trying to say is that **you** are expected to do your research when you run into an issue like the `setup.exe` of a feature upgrade reporting that your platform can't be upgrade, as you will easily find how to bypass that if needed (which, again, you can simply do by using Rufus to create a drive and then running `setup.exe` from that drive).

Rufus is not intended as a "hand holding" application. If that is what you want, please use a different utility, because I am **not** going to add what you request, on account that it is both unrealistic to expect that people will remember, MONTHS OR YEARS down the line, that they got a warning telling them that they may have to work around stuff to install feature upgrades (heck, I'm well placed to know that people do NOT remember a prompt that they got a few minutes ago telling them to try DD mode in Rufus if they encounter an issue with ISO mode, and user forums are FULL of issues reported by people who would have themselves sorted out **if** they had just paid attention to a VERY PROMINENT WARNING FROM RUFUS), and, even outside of this, that the cost/benefit of adding such a warning is too prohibitive.

NOT. GONNA. HAPPEN.