pbatard/rufus #2229: Installation source error with RHEL 9.1 ISO

Naranthiran, 2023-04-19:
﻿

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ x] The log I am copying is the FULL log, starting with the line `Rufus version: 3.22.2009` - I have NOT removed any part of it.

Additionally (if applicable):
- [x ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ x] I also tried one or more of the following:
  - [ x] Using a different USB drive.
  - [x ] Plugging the USB into a different port.
  - [x ] Running Rufus on a different computer.
- [ x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I am rufus tool to create a bootable USB Linux installation media(RHEL 9.1)
The installation get stuck with the error "installation source warning"
The same ISO image works fine with the DD command and balenaEtcher.
Log
---
```
Rufus x86 v3.22.2009
Windows version: Windows 10 Enterprise, 64-bit (Build 19044.2846)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
0 devices found
Found USB 3.0 device 'Apricorn Secure Key 3.0 USB Device' (0984:1410)
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7602, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {67E3004E-7A04-4E6E-9386-D47DE670E86B}
Max parts: 128, Start Offset: 17408, Usable = 62535990272 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: FAT32
  ID: {C95CA8C4-5B64-48F6-9F30-13258EF5D41B}
  Size: 58.2 GB (62534959104 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Found USB 3.0 device 'Apricorn Secure Key 3.0 USB Device' (0984:1410)
Could not open '\\?\usbstor#disk&ven_apricorn&prod_secure_key_3.0&rev_0444#116120018420&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
0 devices found
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'SMAVM'
  Size: 1.8 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: Philips-SGS-RHEL-9-1-230407.iso (1.8 GB)
Found USB 3.0 device 'Apricorn Secure Key 3.0 USB Device' (0984:1410)
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7602, Tracks per cylinder: 255, Sectors per track: 63
[... 124 lines cut ...]
Extracting: D:\isolinux\splash.png (186 bytes)
Extracting: D:\isolinux\vesamenu.c32 (26.1 KB)
Extracting: D:\isolinux\vmlinuz (11.1 MB)
Created: D:\syslinux.cfg → /isolinux/isolinux.cfg
Finalizing, please wait...
Created: D:autorun.inf
Created: D:autorun.ico

Found USB 3.0 device 'Apricorn Secure Key 3.0 USB Device' (0984:1410)
Using 'autorun.inf' label for drive D: 'SMAVM'
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7602, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {27803CCE-7EAB-4834-B48C-5B416A6F4390}
Max parts: 128, Start Offset: 17408, Usable = 62535990272 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: FAT32
  ID: {2F900463-3D61-42D0-8E69-A196730C69E1}
  Size: 58.2 GB (62534959104 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000

```

pbatard, 2023-04-20:
> but I am using RHEL 9.1 ISO.

You are not. You are using a _custom_ ISO that is based on RHEL 9.1, but there is no official RHEL 9.1 ISO that I know that is only 1.8 GB in size and that is labelled `SMAVM`.

>  I can see changing from inst.stage2 to inst.repo works.

It does when you are using an official RHEL 9.1 image, which you are not. I'm afraid I will not help you with __custom__ ISOs, especially as you should be able to just use DD mode, because there are just too many ways you can have deviated from the standard ISO when it comes to config files, and I don't time to invest in trying to help with non public ISOs.

So you really have 3 options here:
1. Use the official RHEL 9.1 ISO.
2. Use DD mode.
3. Sort out your configuration issues entirely on your own.

Naranthiran, 2023-04-20:
Hi Pete,

Thanks for your support!

I was not getting the ISOHybrid image (Write in DD image mode) earlier, as the isohybrid command was not executed on the ISO created.

I was able to solve the "Installation source error" by using the option "Write in DD image mode".

Regards
Naranthiran Durisamy