pbatard/rufus #2230: rufus trys to make a mbr uefi non-csm usb

solomoncyj, 2023-04-21:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Rufus tried to make a MBR Drive but made it boot with UEFI non-CSM only
[image]

iso details
Acronis Bootable Rescue media - Lunix based
SHA256: 68b3aecc3b49c96d72b422a36f644cbf00d31015a0241196c3f5d8c9a0ecb31c

Log
---
```
<FULL LOG>
Rufus x86 v3.22.2009
Windows version: Windows 10 Pro, 64-bit (Build 19045.2846)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'WD My Passport 25E1 USB Device' (1058:25E1)
Device eliminated because it was detected as a Hard Drive (score 43 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Scanning image...
ISO analysis:
  Image is an ISO9660 image
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'ACRONIS_MEDIA'
  Size: 448.3 MB (Projected)
  Uses: EFI
Using image: AcronisBootableMedia.iso (685.3 MB)
Found USB 3.0 device 'WD My Passport 25E1 USB Device' (1058:25E1)
1 device found
Disk type: FIXED, Disk size: 2TB, Sector size: 512 bytes
Cylinders: 243197, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {58E44C8C-F771-4435-AAEA-3ED65A7E9CE0}
Max parts: 128, Start Offset: 17408, Usable = 2000365255168 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'My Passport'
  Detected File System: NTFS
  ID: {5B389A93-B9F5-430F-BD91-409DFCF1D622}
  Size: 1.8 TB (2000362143744 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Checking for Rufus updates...
Checking release channel...
No new release version found.

```

pbatard, 2023-04-22:
```
Disk image analysis:
  Image has an unknown Master Boot Record
  (...)
  Uses: EFI
```

See https://github.com/pbatard/rufus/wiki/FAQ#im-trying-to-work-with-an-iso-but-rufus-says-its-unsupported.

Basically, the Acronis folks decided to use a **custom** BIOS boot method, and since everybody on the planet can invent a custom BIOS boot method, I just don't have the scope to support them all in Rufus (because each BIOS method needs a custom conversion for USB boot).

As such, I have **no** plans to look into adding custom support for Acronis BIOS boot, especially as I believe you should be prompted by Rufus to choose between writing the image in ISO mode or in DD mode, and, when choosing DD mode, you should get what you want.

pbatard, 2023-04-22:
Oh, and just in case this is your "issue", please be aware that MBR is 100% compatible with UEFI, so it doesn't matter if Rufus picks MBR or GPT to create a UEFI bootable media, as it **WILL** boot exactly the same.

closed 2023-04-22 by pbatard