pbatard/rufus #2232: Error extracting WinPE ISO, Possible regression

chazste, 2023-04-22:
﻿

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
ISO file created using ADK promp with 
```
Makewinpemedia /iso /f winpe_x64 winpe.iso
```

This file writes OK with rufus-3.11.exe, fails with 3.17, 3.21 and 3.22

Log
---
```
Rufus x86 v3.22.2009
Windows version: Windows 10 Enterprise, 64-bit (Build 19041.1415)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6700)
Using 'autorun.inf' label for drive D: 'DVD_ROM'
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3773, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {7661BD53-D66A-4B76-86E2-38939F8DA134}
Max parts: 128, Start Offset: 17408, Usable = 31042009600 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {D0BF7E29-F73D-4431-8E60-A90CEE628C1B}
  Size: 28.9 GB (31039884288 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {ACF9C5CA-1267-45B6-913C-E3D16F2372FD}
  Size: 512 KB (524288 bytes)
  Start Sector: 60626822, Attributes: 0x0000000000000000
Could not open Internet session: Internet is disconnected.

Image provided: 'winpe.iso'
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'DVD_ROM'
  Size: 342.7 MB (Projected)
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
Using image: winpe.iso (345.9 MB)

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 28.9 GB)
● Creating UEFI:NTFS Partition (offset: 31040932864, size: 1 MB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Formatting to NTFS (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Using Rufus protective MBR
Writing protective message SBR
Found volume \\?\Volume{f43a2f79-e0e1-11ed-88ac-80000b811aa4}\
Disabling file indexing...
Successfully remounted \\?\Volume{f43a2f79-e0e1-11ed-88ac-80000b811aa4}\ as F:
Extracting files...
libcdio: could not retrieve file info for `winpe.iso': No such file or directory
libcdio: could not retrieve file info for `winpe.iso': No such file or directory
'winpe.iso' doesn't look like an ISO image

Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6700)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3773, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {3938BA55-1116-4E77-8541-A1B46FE44099}
Max parts: 128, Start Offset: 17408, Usable = 31042009600 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {C5AF40B0-90C4-4317-8A46-9F54E4DB24A1}
  Size: 28.9 GB (31039884288 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {E90680DE-A2A3-454D-ADE6-64A37530D213}
  Size: 1 MB (1048576 bytes)
  Start Sector: 60626822, Attributes: 0x9000000000000000

```

pbatard, 2023-04-22:
Not seeing any issue here running on Windows 11:

1. Installed Windows ADK (through `adksetup.exe`).
2. Installed Windows ADK Windows Preinstallation Environment Add-ons (through `adkwinpesetup.exe`).
3. Open ADK command prompt (_Windows Kits_ &rarr; _Deployment and Imaging Tools Environment_).
4. Ran `copype amd64 C:\winpe_x64` which completed successfully.
5. Ran `MakeWinPEMedia /iso /f C:\winpe_x64 C:\Downloads\winpe.iso` which completed successfully.
6. Wrote the image with Rufus:
```[Sat Apr 22 22:01:00 2023]
Rufus x86 v3.22.2009
Windows version: Windows 11 Pro for Workstations, 64-bit (Build 22621.1555)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'Mushkin Ventura Ultra USB Device' (174C:55AA)
Using 'autorun.inf' label for drive F: 'Ubuntu 23.04 amd64'
1 device found
Disk type: FIXED, Disk size: 120 GB, Sector size: 512 bytes
Cylinders: 14593, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x0056DADD
Drive has a Grub 2.0 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 91.8 GB (98558240256 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: ext3
  Size: 20.0 GB (21474819072 bytes)
  Start Sector: 192498611, Boot: No
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'DVD_ROM'
  Size: 337.1 MB (Projected)
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
Using image: winpe.iso (340.3 MB)
Local time: 13326670875
Next update check in 1801 seconds.

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Opened \\.\PhysicalDrive9 for exclusive write access
Analyzing existing boot records...
Drive has a Grub 2.0 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 111.8 GB)
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened \\?\Volume{a8690ef1-6108-4540-b191-d927c9065afd} for exclusive write access
Size : 111.8 GB 234439567 sectors
Cluster size 32768 bytes, 512 bytes per sector
Volume ID is 1de8:1d52
32 Reserved sectors, 28612 sectors per FAT, 2 FATs
3662223 Total clusters
3662222 Free clusters
Clearing out 57320 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
[... 203 lines cut ...]
Extracting: F:\sv-se\bootmgr.efi.mui (78.9 KB)
Extracting: F:\tr-tr\bootmgr.efi.mui (78.3 KB)
Extracting: F:\uk-ua\bootmgr.efi.mui (79.9 KB)
Extracting: F:\zh-cn\bootmgr.efi.mui (64.4 KB)
Extracting: F:\zh-tw\bootmgr.efi.mui (64.3 KB)
Finalizing, please wait...
Created: F:autorun.inf
Created: F:autorun.ico

Found USB 3.0 device 'Mushkin Ventura Ultra USB Device' (174C:55AA)
Using 'autorun.inf' label for drive F: 'DVD_ROM'
1 device found
Disk type: FIXED, Disk size: 120 GB, Sector size: 512 bytes
Cylinders: 14593, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {79D40690-8927-4F0E-B886-A984530A2DFA}
Max parts: 128, Start Offset: 17408, Usable = 120034089472 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: FAT32
  ID: {A8690EF1-6108-4540-B191-D927C9065AFD}
  Size: 111.8 GB (120033058304 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
```

pbatard, 2023-04-22:
Oh and just in case you wanna point that out, I also tried in UEFI:NTFS mode by selecting GPT/NTFS and saw no issue.

pbatard, 2023-04-22:
Ah, so you **are** using commandline to specify the ISO. You should definitely have mentioned that, because your issue has nothing to do with WinPE.

You just need to provide the full path for your ISO if you use commandline as Rufus is not guaranteed to look in the same directory as the executable. Also note that commandline is not really supported. It is provided as a courtesy, but I make no promise about its reliability... or even its ability to work at all.

chazste, 2023-04-23:
@pbatard 
Sorry about that. Good thing the log tells the whole story well. I can confirm that specifying the full path to the ISO does work with 3.22. Thanks! I am OK if you close this. However I will make the following observations:
- The error does not look like "file not found". As a developer I likely would have figured that out.
- It used to work (in 3.11)
- The check button to compute checksums does find the file and works as expected.

closed 2023-04-23 by pbatard

chazste, 2023-04-23:
@pbatard 
Thank you for this change. With this change it is now clear that rufus when does not find the ISO file. However I disagree that a relative path for the ISO file is relative to app_dir. I would expect is to be relative to CWD (i.e. the directory returned by GetCurrentDirectory() called as the first line of WinMain()) This is the directory specified in the "Start in" control of a shortcut to rufus or in my case the directory displayed in the prompt string of my CMD window. I usually download rufus to c:\bin which is in my path and then say
```
rufus-3.22 -g -i test.iso
```
From my testing I would happy with just dropping the following line:
```
IGNORE_RETVAL(_chdirU(app_dir));
```
especially since other places in rufus.c you are very careful to match calls to _chdirU (like pushd/popd)

Also it would nice if the file not found printf went to the log as well. Maybe that should be a ubprintf? That line appears to be the only bare printf in rufus.c other than the help message.

pbatard, 2023-04-23:
Sorry, but that's all you're going to get for now. The time I have to devote on commandline support is very limited and I have other things to take care of. Please use a fully qualified path if you are still unhappy about the current fix.

pbatard, 2023-04-24:
OK, I removed the `_chdirU()` in 1e51385bedcef1ce94ed6c2f52b25e04ef8a0508. But I haven't fully tested it, so if it happens to breaks something else, so be it.

chazste, 2023-04-25:
Thanks again!

commit a3541782 (2023-04-23): [cmd] fix the use of relative paths with option -i

* Closes #2232