pbatard/rufus #2234: i miss a version. sorry.

zedd-0, 2023-04-26:
is it possible to get the microsoft store version of 'rufus v3.22' ?
i forgot to download it, and on github, there is no *.appx-version available.

(i use the appx-versions, because they are not compressed*, and it has a x86-64 version)
(*this compress better for my archive...)

pbatard, 2023-04-26:
> is it possible to get the microsoft store version of 'rufus v3.22' ?

No.

> and it has a x86-64 version

You do realize that, per the homepage, `rufus-4.0.exe` is now an x86_64 version right?

At any rate, I don't support older versions. The only version I can support is Rufus 4.0.

closed 2023-04-26 by pbatard

zedd-0, 2023-04-26:
sorry for the misunderstanding.
i search it for the version 3.22

pbatard, 2023-04-26:
> i search it for the version 3.22

The **only** 3.22 versions you will find are the ones provided at https://rufus.ie/downloads/. That's all.