pbatard/rufus #2235: Windows could not apply unattend settings

SegoItCh, 2023-04-26:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Getting Windows could not apply unattend settings after install with latest Windows.iso and this settings
[image]
>

SegoItCh, 2023-04-26:
Keeping only first two checks(default) made it work

pbatard, 2023-04-26:
No log &rarr; closed. I cannot help you if you decide not to provide the log as requested.

closed 2023-04-26 by pbatard

SegoItCh, 2023-04-26:
I don't need help, I reported issue, you need log of burning latest Windows.iso to flash? 
I fixed problem for myself by turning off latest two checks in Windows User Experience window

SegoItCh, 2023-04-26:
@pbatard

SegoItCh, 2023-04-26:
Latest that you get from mediacreationtool

pbatard, 2023-04-27:
Can't replicate your issue.

Downloaded the latest Windows 11 International English ISO using MCT
Created a bootable drive using Rufus 4.0 and the same WUE options you selected
Installed Windows Pro on a machine that never had Windows installed (thereby allowing me to select the version I wanted) and I did not see any error.

Maybe there are some elements I needed to be able to replicate your issue, some of which I may have been able to pick from the log if you had provided one, but if I were to venture a guess I'd say that you have been trying to install Windows 11 on a machine that had already been activated for a Windows _S Edition_ (because Windows does look for the hardware fingerprint to select the edition of Windows it should install), and, because _S Editions_ are a lot more restrictive than regular Windows versions, some of the unattend settings could not be applied.

However, in the absence of comprehensive data with regards to your environment, target and other elements, that's a complete guess, and since I cannot replicate your issue in a _clean_ hardware environment, there's nothing more I can do to investigate this issue.