pbatard/rufus #2236: Rufus Portable can't write to the rufus.ini file when it has a 'hidden' attribute.

ALIENQuake, 2023-04-26:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Issue description
-----------------
Rufus Portable can't write to the rufus.ini file when it has a Windows 'hidden' attribute.

Log
---
```
Rufus x64 v4.0.2035 (Portable)
Windows version: Windows 11 Enterprise x64 (Build 22621.1635)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0415 (pl-PL)
Will use default UI locale 0x0415
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'pl-PL'
0 devices found
Checking for Rufus updates...
Checking release channel...
Could not write 'F:\Image\rufus.ini' - original file has been left unmodified
No new release version found.

```

pbatard, 2023-04-26:
I'm afraid that I don't see enough people having this issue to want to do anything about it.

This only happens if someone **chooses** to deliberately set the hidden attribute.

If you don't want to see a `.ini` file, then use the non portable version, especially as I am willing to bet that your idea of what the Rufus portable version does (not write to the registry ever) and what it actually does (it **does** write to the registry, even if it's labelled "portable" [because portable does not imply anything about leaving the registry alone](https://github.com/pbatard/rufus/wiki/FAQ#user-content-What_is_the_difference_between_the_portable_and_the_regular_version)) are most likely not the same...

closed 2023-04-26 by pbatard

ALIENQuake, 2023-04-26:
@ALIENQuake Yes, I choose to deliberately set the hidden attribute, its basic OS function. What's the problem? I never face a similar issue in 20 years of using an app. This is definitely a bug.

I don't really care about confusion regarding the 'what's portable actually means' thing. I'm perfectly fine with Rufus writing to the registry as long as I don't need to install the app after reinstalling Windows.

pbatard, 2023-04-26:
> What's the problem?

The problem is that only one person (you) is likely to ever be affected by this issue, and I am not going to go out of my way to spend time addressing something that only one person is ever likely to face and especially something that can easily be remedied (don't set the hidden attribute).

The cost/benefit of looking into this is simply not worth it.

> I'm perfectly fine with Rufus writing to the registry as long as I don't need to install the app after reinstalling Windows.

Then you can use the regular version, since Rufus does not come with an installer.

ALIENQuake, 2023-04-26:
Alright, I won't nag you about this. I don't know c++ so I can't contribute. I'm adding this in case of someone wants to do PR: https://learn.microsoft.com/en-us/windows/win32/api/fileapi/nf-fileapi-createfilea?redirectedfrom=MSDN#files
> If CREATE_ALWAYS and FILE_ATTRIBUTE_NORMAL are specified, CreateFile fails and sets the last error to ERROR_ACCESS_DENIED if the file exists and has the FILE_ATTRIBUTE_HIDDEN or FILE_ATTRIBUTE_SYSTEM attribute. To avoid the error, specify the same attributes as the existing file.