pbatard/rufus #2237: Threat detected: HEUR/APC

1LUC1D4710N, 2023-04-26:
The latest version 4.0 is being removed automatically when it logs threat detected: HEUR/APC

Source: 

https://objects.githubusercontent.com/github-production-release-asset-2e65be/2810292/db855312-b398-4e85-a0dd-952b3702d5be?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230426%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230426T201517Z&X-Amz-Expires=300&X-Amz-Signature=cedc11d7227da9ab04edf795bb85c6e385e6e4f149ead0ee21d474d139507f01&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=2810292&response-content-disposition=attachment%3B%20filename%3Drufus-4.0p.exe&response-content-type=application%2Foctet-stream

pbatard, 2023-04-26:
What anvirus are you using?
What version of Rufus are you using (x64, x86, ARM)?

For reference [THIS](https://www.virustotal.com/gui/file/bfecf4dcf1a63d8b64b900906102edf666642316291c9bba42eb0fb9c7bccbd6/detection) is the current VirusTotal detection status for Rufus 4.0 (x86_64, i.e the main executable being proposed), and none of the major AVs are flagging a false positive.

Also, [per the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Antivirus_X_reports_that_Rufus_contains_malware), I will kindly ask that you report this false positive to your AV vendor, as, unless it's a major one (like Microsoft Defender) I genuinely don't have the time to go around reporting yet another false positive to the MANY poorly designed AV engines that find nothing better than report Rufus as malware every other day.

1LUC1D4710N, 2023-04-26:
Threat protection and VirusTotal indicate it as such, my AV didn't even
detect it.

***@***.***

pbatard, 2023-04-26:
Is `Threat protection` the name of an antivirus? I can't seem to find any hit for it. If not, you have to provide the name of the company that makes your antivirus.

And, as I pointed out, VirusTotal has its **usual** 1 outlier false positive (I can't remember ever releasing a version of Rufus where one of the many AV VirusTotal lists would not report a false positive).

Again, because I don't have the actual manufacturer of your security solution, there's literally nothing I can do about your report. And, because none of the major AV vendors are reporting that threat, I will kindly ask you to take care of it yourself, by getting in touch with your AV vendor.

closed 2023-04-26 by pbatard

1LUC1D4710N, 2023-04-26:
Threat protection is part of NordVN for their AV protection 🙂
Like Surfshark has Surfshark One as part of their AV protection.

I am using Bitfender as AV, here it didn't have the same reaction.

***@***.***

pbatard, 2023-04-26:
> Threat protection is part of NordVN for their AV protection

Then report the false positive to NordVPN.