pbatard/rufus #2238: Rufus 4.0 stucked on 96.5%

YashvirGaming, 2023-04-29:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Hi Rufus latest version v4.0 seems to be stuck at 96% 10mins passed by nothing.
I already cancelled and re do the whole thing again it stucked here on 96%
previous version 3.22 worked like a charm. 
Please fix your v4.0 latest. 
[image]

Log
---
```
<FULL LOG>
```

YashvirGaming, 2023-04-29:
Oups sorry its done now it took 13mins total on my machine RTX 4080, 32gb , ryzen 5800x3d
Please close this ticket 

[image]

closed 2023-04-29 by YashvirGaming

pbatard, 2023-04-29:
Yeah, writing times may fluctuate depending on what's happening with your environment, or the USB media, so please be careful of not assuming that some behaviour you only saw once or twice is a regression, especially if nobody else appears to be reporting the same thing (because if there was a regression with regards to the creation of Windows 10 drives using Rufus, quite a few people would have reported it already).