pbatard/rufus #2241: Windows 11 insider Dev indicates that I can no longer download a cumulative update because my PC doesn't meet its requirements.

Dhyfer1, 2023-05-06:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I 'm a Windows 11 insider user in the Dev channel and this is the first time I've encountered a problem that may or may not be related to Rufus.

Before using Windows 11 Dev channel, I was a beta channel user. I download the Windows 11 insider beta channel ISO file from the [official website](https://www.microsoft.com/en-us/software-download/windowsinsiderpreviewiso), this iso file was mounted with Rufus to a usb flash drive leaving checked (before flashing) only the first option:

`✅Eliminate the requirement of more than 4GB of RAM, Secure Boot and TPM 2.0`

When rufus successfully finished the flashing process, I installed Windows 11 beta channel from the BIOS. And when Windows 11 beta finished installing, Windows Update could download and install all updates without any problem.

Now I have a problem with the dev channel. I have done the same steps as described above, but this time using the ISO file for the dev channel. After Windows 11 dev finished installing, I check for updates in Windows Update and I get the cumulative update you see in the image, which increases my build number from 23440.1000 to 23451.1000.

When Windows Update reaches 8% downloading this update, this dialog box appears indicating that my PC doesn't meet the Secure Boot and TPM requirements. And when I close this dialog box, Windows Update does not continue downloading, instead it clears whatever it has downloaded up to that 8%, and the cumulative update instead of showing Download, it just shows Retry.

[image]

I have retried Windows Update to download this cumulative update, but always at 8% of download this dialog box appears. So I do not understand, as I said before I have checked the option to remove the secure boot requirement and TPM in rufus, but for these requirements Windows Update doesn't let me continue with the download. This is the first time this has happened to me, I have never had this dialog box appear in Windows 11 beta channel and I have never had problems with Windows Update in the beta channel.

My problem occurs when I use the latest version of rufus on my unsupported PC running Windows 11 Pro, 64 bit architecture, Windows insider Dev channel.

pineapple63, 2023-05-07:
The option to bypass secure boot and TPM only applies during the installation

Updates which only increment the second part of the build number will install without issue (unless Microsoft change something in the future), but updates which increment the first part of the build number will re-run the compatibility check (for the release version of Windows 11, the only time this would be an issue is when upgrading from 21H2 to 22H2, or 22H2 to 23H2, but the dev and canary insider channels increment the first part of the build number more often, and they don’t always release ISOs for the new builds)

pbatard, 2023-05-07:
Indeed, the bypasses applied by Rufus are __not__ persisted to the OS, as they only apply for the specific installation you carry out with the media created by Rufus, so once you have installed Windows 11 using the bypasses, you will have to find some other way to work around whatever limitations Microsoft want to impose.

The obvious reason for this is that it would be very poor practice for Rufus to modify OS binaries during installation (we do want to provide some assurance that if someone installs Windows through Rufus, they end up with the __exact same OS files as if they had installed Windows through the official means__), and with most of the binaries being signed, our only choice would probably be to install a permanent service that hijacks system calls or something... which isn't something that users worried about how much they can trust the Rufus application would be too happy about. Plus, whatever solution we'd implement would be at the mercy of Microsoft adding an update that breaks it, and it'd be a cat and mouse game where we always lose.

So, we consider that once the OS has been installed, and even if Rufus has helped you work around the TPM or Secure Boot bypasses, it is up to **you** to figure out how you can continue to apply these bypasses from within the installed OS.

For instance, have you tried setting the various `HKEY_LOCAL_MACHINE\SYSTEM\Setup\LabConfig\`  keys to see if that helped.
Or have you tried locating the `appraiserres.dll` that is most likely being used by the update check, and replacing it with a 0-byte file?

Dhyfer1, 2023-05-08:
@pineapple63 Thank you very much for the clarification, I have understood correctly. So rufus only allows me to skip these requirements when I install Windows 11, and since I am running Windows 11 Dev (channel that increments the first part of the compilation number more often, as you have said), any updates that check compatibility will not be able to download until I find another way to skip these requirements when the OS is currently installed.

> ...but updates which increment the first part of the build number will re-run the compatibility check

I didn't know this, thanks for the info. That seems to be the case with the update that (because of not meeting the requirements) couldn't increase the build number from 23440.1000 to 23451.1000. As you can see this update increases the first part of the build number, but before it reaches 100% download it re-runs the compatibility check.

@pbatard Tambien agradezco tu aclaracion. Crei que los cambios hechos por Rufus para deshabilitar TPM y arranque seguro eran permanentes, bueno, parece que los cambios si son permanentes para el canal beta ya que como dije antes nunca tuve problemas con Windows Update, siempre podia descargar cualquier actualizacion sin que saltara este cuadro de dialogo diciendome que mi PC no cumple los requisitos. En cambio, el canal Dev (y es posible que el canal Canary tambien) son canales muy estrictos que verifican la compatibilidad antes de aplicar cualquier actualizacion, asi que parece que depende del canal insider de Windows 11 que se este usando.

>So, we consider that once the OS has been installed, and even if Rufus has helped you work around the TPM or Secure Boot bypasses, it is up to you to figure out how you can continue to apply these bypasses from within the installed OS.

You are right. Rufus only does one part of the job by helping me install Windows 11 on my unsupported PC, and if Windows Update tells me I don't meet the requirements then I have to find another way to bypass these requirements when the OS is installed.

>For instance, have you tried setting the various HKEY_LOCAL_MACHINE\SYSTEM\Setup\LabConfig\ keys to see if that helped.
Or have you tried locating the appraiserres.dll that is most likely being used by the update check, and replacing it with a 0-byte file?

I have not tried the registry editing, but I have tried replacing the appraiserres.dll file and thanks to that I was able to download and install the latest build of the dev channel and Windows Update gave no more problems. Some say that it is easier to follow this method of replacing this file than using the script [Skip_TPM_Check_on_Dynamic_Update.cmd](https://github.com/AveYo/MediaCreationTool.bat/blob/main/bypass11/Skip_TPM_Check_on_Dynamic_Update.cmd). I also found out on internet that the appraiserres file must be extracted from the Windows 10 ISO file and replaced with the same file found in C:\$WINDOWS.~BT\Sources in Windows 11, I thought that the appraiserres file found on the internet comes from any PC that does meet all the requirements to install Windows 11, but it doesn't.

pbatard, 2023-05-09:
OK, I'm going to close this issue then.

closed 2023-05-09 by pbatard