pbatard/rufus #2247: version rufus 4.0 does not recognize usb on the contrary previous versions

oswram19, 2023-05-13:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
I have a problem that the new version of rufus does not recognize a certain usb, contrary to a previous version if it recognizes it, version 3.18 recognizes it, and the new version 4.0 does not
Log
---
```

<FULL LOG>
```

oswram19, 2023-05-13:
I have a problem that the new version of rufus does not recognize a certain usb, contrary to a previous version if it recognizes it, version 3.18 recognizes it, and the new version 4.0 does not

VitaSmith, 2023-05-13:
Can't help you without a log. Version 4.0 __does__ provide a log if you follow the steps provided in the check list, I guarantee it.

Unless you provide a full log, I will close this issue.

oswram19, 2023-05-13:
This would be the one that worked for me until version 3.21 worked fine
Rufus x86 v3.21.1949 (Portable)
Windows version: Windows 11 Pro, 64-bit (Build 22621.1702)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x080A (es-MX)
Will use default UI locale 0x080A
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'es-ES'
Found USB 2.0 device 'SCSI DISK USB Device' (23A9:EF18)
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 981, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {4F0F0214-E29F-4144-8AD1-1C4B5E6E60CC}
Max parts: 128, Start Offset: 17408, Usable = 8069855744 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {D47CD5AC-1936-41E9-861F-EAB81046F014}
  Size: 7.5 GB (8067741696 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {176650B4-E6B0-4DB1-B9EC-9A1793CA4DEE}
  Size: 1 MB (1048576 bytes)
  Start Sector: 15759356, Attributes: 0x9000000000000000

oswram19, 2023-05-13:
Rufus x64 v4.0.2035 (Portable)
Windows version: Windows 11 Pro x64 (Build 22621.1702)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x080A (es-MX)
Will use default UI locale 0x080A
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'es-ES'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'SCSI DISK USB Device' (23A9:EF18)
Device eliminated because it was detected as a Hard Drive (score 5 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found

oswram19, 2023-05-13:
> Can't help you without a log. Version 4.0 **does** provide a log if you follow the steps provided in the check list, I guarantee it.
> 
> Unless you provide a full log, I will close this issue.

Rufus x64 v4.0.2035 (Portable)
Windows version: Windows 11 Pro x64 (Build 22621.1702)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x080A (es-MX)
Will use default UI locale 0x080A
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'es-ES'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'SCSI DISK USB Device' (23A9:EF18)
Device eliminated because it was detected as a Hard Drive (score 5 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found

pbatard, 2023-05-14:
Can you please describe the manufacturer and model name of your USB device, as well as its capacity?

A link to where your purchased it (or an equivalent page) would be a plus.

Please be mindful that Rufus is designed __NOT__ to list external USB HDDs by default (or similar devices by default), and I regularly update the code to make sure that this is the case (i.e. USD HDDs that were __erroneously__ listed by default will be removed). So I need to know the exact nature or your device, along with additional details, to decide whether I want to add an exception.

Especially, a device that identifies itself simply as `SCSI DISK USB Device` is usually an external HDD rather than a flash drive, so I need to make sure that it is indeed a flash drive before I can add an exception for it.

pbatard, 2023-05-14:
Also note that you can __always__ list USB drives by following [this entry from the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Help_my_USB_drive_is_not_detected), so you are not exactly stuck here.

commit 109df708 (2023-05-29): [misc] add an exception for 23A9:EF18 UFDs

* Closes #2247
closed 2023-05-29 by pbatard