pbatard/rufus #2248: Status of "READY" is meaningless

mr-brunes, 2023-05-14:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
When starting Rufus, the Status bar says "READY", even if there is no device or image selected, so it is unclear what exactly is "ready".
I setup and run the job and return later. The Status bar says "READY".
It should say something different to indicate successful job completion.

Log
---
```
<FULL LOG>
```

pbatard, 2023-05-14:
READY has been a staple of status messages since the very first computers. It indicates that the application has completed what it was doing and is __ready__ for a new task.

I have no plan to change that message.

closed 2023-05-14 by pbatard

mr-brunes, 2023-05-15:
That's a shame. I don't recall the early computers as being very easy to use, as they focused on internals and not users.

Wording this a different way, I don't see how one can check on return that a job has run successfully since the status is READY at the first invocation of the app, after configuring a job and afterwards. I checked the log file but couldn't spot a "Task completed successfully" or somesuch entry.
Maybe Status actually means "App Status" when I'm looking for a  "Job Status". Feature request?

pbatard, 2023-05-15:
> I don't recall the early computers as being very easy to use

I beg to differ. Early computers didn't bombard the user with tons of different label and message cues to say the same thing, just because the interface designer found it "prettier that way".

>  I don't see how one can check on return that a job has run successfully 

Progress bar showing 100% not enough?

Or do you really believe that applications are so poorly designed that if they detect an error or any issue that prevented successful completion, they will not report it to the user?
Your mistrust of the design of the application is your downfall here, because if you don't get an error and the application says it's ready, __then__ it means that the task it was performing completed successfully.

Therefore, I am bound to assume that, if you request a different message than "ready", it's because you are expecting me to have done a bad job when developing the application, by allowing scenarios where an error would be detected and not reported the user.

Well, as you should expect, I don't design applications for people who seem to expect that I am a lousy software developer and that I will somehow allow a job not to complete without reporting it VERY EXPLICITLY to the user.

mr-brunes, 2023-05-15:
The early ones I recall (just about) didn't have what I call an 'interface design' per se. The lone prompt or flashing  cursors were pretty impenetrable to most, unless one immersed oneself in the OS manual (remember those?). ;-)

As for mistrust in the app or developer, you couldn't be more wrong. It's about mistrust of my own memory!
In other words, I *think* I started an image write but got distracted and now I can't recall. I'm an occasional user and each time I use Rufus, upon which I settled *because* of the reliability) I find myself thinking the same thing, and yesterday it just struck me what would help. Apols if I worded the request inconsiderately.
FWIW I can't see a 100% indicator.

pbatard, 2023-05-15:
> FWIW I can't see a 100% indicator.

A green progress bar filled is a 100% indicator. And your distraction is not something for a developer to address.

mr-brunes, 2023-05-16:
FWIW I'm colour blind so the green aspect completely passed me by. Using colours alone as a differentiator is not recommended as per many UX design guides. Adding a simple "100%" text label would help make a difference and largely solve this issue.
As for my distraction, I hope sincerely that you are not affected by any sensory or cognitive differences in your life. I would note that a growing number of developers do now consider these things in their UX designs.

pbatard, 2023-05-16:
While Rufus is performing its operations, the progress bar also displays a dumber percentage that goes from 0% to 100%. So, the only problem I see is that you are starting an operation, and looking completely away from the interface (else you would see the progress being reported both as a filling shape clue and a text clue) and then come back to Rufus not knowing if you clicked before you looked away.

I'm sorry, but that is not typical behaviour.

1. Someone starting an operation would stay at its onset, especially as you will be prompted if you really want to erase the drive (i.e. you can't just click once and go) with a dialog prompt. Before and after that prompt, the text on the progress bar will have changed to something else than "READY" telling you that the application is actively doing something.
2. To be even remotely confused as to whether Rufus did or did not complete an operation that you previously started, you'd have to come back, having completely forgotten the following:
   - That you __actively__ clicked START.
   - That you __actively__ clicked to accept for your drive to be erased.
   - That you saw the text from the status bar change to an active (non "READY" status), and, most likely, saw a percentage numbers indicating the start of the progress towards 100%.

Oh, and none of the above will be different whether you are colourblind or not.

You may be colourblind, but it also does look to me that you __are__ inventing excuses that don't really stand the test of scrutiny to try get your way. Rufus does provide more than enough cues for the "READY" status to be 100% meaningful.