pbatard/rufus #2251: Using Rufus 3.2 and 4.0 getting error messaging to try again later

Sunset439, 2023-05-18:
[image]

Sunset439, 2023-05-18:
> Yes, it appears that Microsoft removed access to older versions of
> Windows, so if you attempt to download anything but the latest release of
> Windows 10 or Windows 11, you will get this error.
>
> This is being tracked in pbatard/Fido#67
>
> Obviously, we can't server ISOs that Microsoft servers no longer want to
> serve, so if these ISOs remain unavailable, I will remove them from the
> list.

Oh no! Have a Toshiba Satellite and need to re install win 10 21H1. No
other way to do this?

pbatard, 2023-05-18:
You might be able to reconstruct an ISO from https://uupdump.net/ but this will take a bit more work than just downloading a file.

JonnyTech, 2023-05-18:
https://files.rg-adguard.net/

Sunset439, 2023-05-19:
Hello again

I chatted woth Microsoft about the error code im receiving amd they told me
its prohibited to use Rufus in my location? I'm in the United States and
never had a problem using it before. Can you shed some light on this?
Something doesn't sound right with this

Sunset439, 2023-05-19:
The downloads at rg-adguard.net are torrent files and I don't know how to
use those to do an install.

Sunset439, 2023-05-19:
I tried using Rufus to download win 11 amd the win 10 22h2 and got the same
error message which according to Microsoft the error code, which is
715-123130 , Rufus can no longer be used in the United States??

pbatard, 2023-05-19:
>  amd they told me its prohibited to use Rufus in my location

No surprise here. You've just confirmed what had only been inferred for a while, in that part of the recent countermeasures that Microsoft applied to their ISO download services was because they were annoyed that Rufus was providing a service that most users want, but that Microsoft doesn't want to provide (i.e. straightforward ISO download without having to jump through hoops).

I guess their discontinuing of providing olders ISOs is a continuation of this...

> I tried using Rufus to download win 11 amd the win 10 22h2 and got the same error message

No issue from me here in Ireland, trying to download the latest Windows 11 and Windows 10 22H2.

Please bear in mind that, if you try a few too many downloads in rapid succession (3 or 4 seems enough to trigger it) the Microsoft servers may stop replying to subsequent queries and produce the dreaded `` error code (which is a different issue from what we're dealing with here, and [is explained in the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Code_715123130_unable_to_complete_your_request_at_this_time_when_trying_to_download_a_Windows_ISO)).

Sunset439, 2023-05-19:
Ok I'll wait a few days and maybe try again. Thank you. Rufus had worked
for me about a month and a half ago with Win 10 20h2. I installed it and it
worked but then display went white and I found out Microsoft also no longer
supports Intel Graphics Media Accelerator 4500M. So I was trying yo
reinstall it today amd them try to find compatible Intel driver that will
work. But have to get Windows on it first! I don't think it will run 22h2
buy if all else fails ill try that right from Microsoft I suppose. Thanks
for your insight, much appreciated!

pbatard, 2023-05-26:
Closing this as pbatard/Fido#67 has been sorted.

closed 2023-05-26 by pbatard