pbatard/rufus #2252: Windows installer UEFI boot issue with Asus Maximus IV Extreme Z and Rufus 4.0

cela96, 2023-05-19:
﻿
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: 4.0` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Motherboard: Asus Maximus IV Extreme Z
Rufus version: 4.0

Problem: The Asus UEFI firmware refuses to boot Windows installer created in UEFI (GPT) mode using Rufus 4.0. 
However, with Rufus 3.22, the PC successfully boots in Windows installer. 
If the USB drive is created with BIOS (MBR) support it works.

Unfortunately, I had a limited time to use this motherboard, and I cannot conduct further tests. I tried different USB drives, and with Rufus 4.0, it consistently had issues when using UEFI, while it worked fine with BIOS. I'm not exactly sure what the problem could be, but there is definitely something with the new version. It could also be related to the fact that the motherboard is quite old, but I haven't encountered any issues with other motherboards of a similar age.

Log
---
```
No log necessary.
```

pbatard, 2023-05-19:
> No log necessary.

Your choice. This is not optional if you want to receive support.

closed 2023-05-19 by pbatard

cela96, 2023-05-19:
OK, sorry. I wrote it that way, but actually I should have said that since I no longer have the motherboard and it was a problem a few days ago, I cannot have that specific log, so I deemed it unnecessary.
What I can do is use the same hardware I used the first time to create a new USB stick with Rufus 4.0 and paste a full log, but I cannot test the "non-functioning" because I no longer have the PC with the Asus Maximus IV Extreme Z motherboard.

pbatard, 2023-05-19:
I need to see how you created the media with Rufus 4.0. There's just too much information I don't have without a log.

cela96, 2023-05-20:
OK, I'll try to get you the logs as soon as possible. The PC I used to create the USB stick is not in my city.

AgentSam, 2023-05-22:
Not to confuse the issue too much, hopefully, but -- I had a similar problem with the latest version of Rufus (4.00):

[https://github.com/pbatard/rufus/releases/tag/v4.0](https://github.com/pbatard/rufus/releases/tag/v4.0)

However, the solution to my particular "_UEFI boot issue_", was simply in the BIOS settings of my _Rampage IV Extreme_ motherboard, wherein I had to configure the following setting:

`Advanced > USB Configuration > EHCI Hand-off = Enabled`

For whatever reason, this allowed UEFI boot (with Secure Boot disabled) to be successful.

Alas, at this time I am unable to provide Rufus logs either. But if I ever work on this motherboard model again, I'll make sure to save any Rufus logs and document the current failed/working BIOS settings in more detail.

But, I can say, with 100% certainty, that the boot did NOT work, unless I **enabled** the `EHCI Hand-off` option in the BIOS.

Perhaps this could provide some clues to the "issue" or perhaps I too will be scolded for not providing the logs. (I do realize how painful it can be to work with insufficient information. But this is the best I can do "right now".)

Here's a screenshot of the BIOS setting, which solved the problem for me:

[image]

pbatard, 2023-05-22:
@AgentSam, I appreciate the help and additional details, thanks. But the one question I have is whether you tried with Rufus 3.22 as well, and if it works then.

In the absence of other information, my current assertion is that you'll find that you had the same issue with Rufus 3.22 with EHCI disabled, in which case what you report seems to be unrelated to what OP was reporting, as OP is claiming that a drive created under the same conditions with Rufus 4.0 works, whereas a drive created with Rufus 3.22 doesn't.

Since EHCI and xHCI hand-off are related to USB handling, it's not entirely surprising that this setting could have an impact as to whether you will be able to successfully boot an installer from a USB drive or not. I would also encourage OP to double check their settings there as well.