pbatard/rufus #2254: Windows Update not working?

turowicz, 2023-05-26:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I have installed Windows 11 with ISO burnt by Rufus with 3 options selected: bypass online account, bypass privacy questions, create local user. After the installation I can't get Windows Update to work. I keep getting error code `0x80080005`. Windows Update Troubleshooter is stuck on "diagnosing". Is this a known issue with Rufus or is this issue specific to my installation?

pbatard, 2023-05-26:
This appears to be specific to your installation.

For one thing, the options used by Rufus are all officially supported Microsoft [unattend installation options](https://learn.microsoft.com/en-us/windows-hardware/customize/desktop/unattend/), meaning that Rufus is not doing anything that could even be remotely considered dodgy or outside of what Windows is designed to support.

Second, none of these options are even remotely close to having a potential effect on Windows updates (even the privacy one, because if updates were only delivered if you rejected the privacy options, Microsoft would have a huge security issue on their hand, alongside very angry Windows users).

Finally, as you can see if you have performed a search, nobody appears to be reporting a similar issue, even though Rufus is downloaded closed to 3 million times each month (and a significant portion of these downloads would be for Windows 11 installations, where users select similar options to you).

The only known limitation I know with Windows update is if you create a _Windows To Go_ drive (which you didn't), but that only applies to Feature Updates, and not to regular updates. Outside of this, the Windows Update service is obviously designed by Microsoft to be able to provide the latest security patches to users, regardless of the options they used during installation (such as using an offline account or enabling privacy), so if that doesn't work, it indicates an environmental problem with your specific platform (e.g. additional software that you installed, and that interferes with Windows Update or a network issue).

turowicz, 2023-05-27:
Thank you for the extensive write up. I'll keep fighting.

closed 2023-05-27 by turowicz