pbatard/rufus #2258: when i was using refuse and

AryanFarhadi, 2023-05-30:
Error : 

Rufus x64 v4.0.2035
Windows version: Windows 10 Pro x64 (Build 19044.1826)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'USB Flash Disk USB Device' (090C:1000)
Ignoring drive '\\.\F:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\F:' as it spans multiple disks (RAID?)
Found USB 2.0 device 'MXT-USB Storage Device USB Device' (AAAA:8816)
Device eliminated because it appears to contain no media
1 device found
Ignoring volume '\\?\Volume{00000001-0000-0000-0000-e0ba0c000000}' because it has more than one extent (RAID?)...
No volume information for drive 0x83
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1937, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 0
Disk ID: 0x00000000
read_sectors: Read error [0x00000015] The device is not ready.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Drive does not have a Boot Marker
Ignoring volume '\\?\Volume{00000001-0000-0000-0000-e0ba0c000000}' because it has more than one extent (RAID?)...
read_sectors: Read error [0x00000015] The device is not ready.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Ignoring drive '\\.\F:' as it spans multiple disks (RAID?)

Format operation started
Requesting disk access...
Ignoring drive '\\.\F:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\F:' as it spans multiple disks (RAID?)
Will use 'Y:' as volume mountpoint
Ignoring volume '\\?\Volume{00000001-0000-0000-0000-e0ba0c000000}' because it has more than one extent (RAID?)...
read_sectors: Read error [0x00000015] The device is not ready.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Opened \\.\PhysicalDrive3 for exclusive write access
Analyzing existing boot records...
read_sectors: Read error [0x00000015] The device is not ready.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Drive does not have a Boot Marker
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Write error [0x00000079] The semaphore timeout period has expired.
Retrying in 5 seconds...
Ignoring drive '\\.\F:' as it spans multiple disks (RAID?)
Write error [0x00000079] The semaphore timeout period has expired.
Retrying in 5 seconds...
Ignoring drive '\\.\F:' as it spans multiple disks (RAID?)
Write error [0x00000079] The semaphore timeout period has expired.
Retrying in 5 seconds...
Ignoring drive '\\.\F:' as it spans multiple disks (RAID?)
Write error [0x00000079] The semaphore timeout period has expired.
Could not reset partitions
Re-mounted volume as Y: after error

Found USB 2.0 device 'USB Flash Disk USB Device' (090C:1000)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'MXT-USB Storage Device USB Device' (AAAA:8816)
Device eliminated because it appears to contain no media
0 devices found

-------------------------------------------------------------------------------------------------------

after failing in make flash bootable I cant create new volume in my flash usb becuse i get error device is not ready

pbatard, 2023-05-30:
Your flash drive happened to fail while Rufus was using it. It wasn't damaged by Rufus.

See https://github.com/pbatard/rufus/wiki/FAQ#user-content-Help_Rufus_damaged_my_flash_drive.

Consumer USB flash-based drives have a much shorter lifespan that people imagine and, if you are unlucky, they will fail much sooner than you would like. They will fail regardless of the application being used and, of course, the application will report an error after that (but, again, you can not assume that the drive failed because of the application).

If you are unlucky, you might also have been sold a "fake" drive, which will fail as soon as you start writing more data than its __actual__ capacity (which, because it is fake will be much smaller than the capacity reported by the drive).

If your drive failed, and it is still under warranty, you should ask the seller for a replacement. There's really nothing else that can be done.

closed 2023-05-30 by pbatard

AryanFarhadi, 2023-05-30:
now what should i do i tried not bootable in rufus now i don't even show the usb flash in disk management

pbatard, 2023-05-30:
> If your drive failed, and it is still under warranty, you should ask the seller for a replacement. There's really nothing else that can be done.