pbatard/rufus #2259: Error 17 no mapping Rufus please help me stuck in this problem for days

Andrewwindowstweaker, 2023-06-03:
Hi recently my windows 11 won't boot from uefi bios even though after selecting it as main os it is still detecting the boot maneger of windows but it ain't boot so I created a windows 11 uefi pendrive from Rufus (disabled secure boot ) and then booted and I got a error (Error 17 no mapping) so I tried another tool named ventoy and when I tried to boot iso it just brings me back to the ventoy menu then I tried official Microsoft media creation tool but no success so I tried windows 8.1 iso and it got installed. I was thinking how windows 8.1 got installed so I tried windows 10 build 1507 iso it also booted but the iso after build 1507 ain't boot in my laptop like 1511 and 1607 and other newer versions. Iam confused why only iso of windows 10 build 1507 and before boots and get installed but if I use latest iso it won't boot . I know this is not the problem of Rufus but I am thinking the dev you can help me please. 

Things I tried : 

Making bios settings to default

Disable secure boot 

Update bios 

Try another isos and tools

(I can boot Linux and windows 10 1507 ) but the newer builds wouldn't boot . I know the usb drive get detected but when I tried to boot from them it literally brings me back to bios 😭

Andrewwindowstweaker, 2023-06-03:
Also my laptop was released in 2021 though it doesn't support CSM or legacy and if u think what the problem is with windows 8 and 10 1507 . The problem is that my laptop drivers won't work with these old versions of windows

pbatard, 2023-06-03:
> but I am thinking the dev you can help me

I'm sorry but you are thinking wrong. I simply don't have the scope to help people on a case by case basis, and it simply looks like recent versions of Windows are incompatible with your system, and there is no "magic trick" that one can use to suddenly make them compatible.

I will therefore close this issue while inviting you to ask for help in __much more__ relevant spaces such as Reddit's [WindowsHelp](https://old.reddit.com/r/WindowsHelp/) of Microsoft's [support pages](https://answers.microsoft.com/en-us).

closed 2023-06-03 by pbatard