pbatard/rufus #2262: Download Button is not visible

iahmtb, 2023-06-10:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
i cannot see the download button even after enabling the check for update option

Log
Rufus x86 v3.21.1949
Windows version: Windows 7 SP1 32 bit
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
Unable to send request: The attempt to connect to the server failed.
Checking for Rufus updates...
Checking release channel...
Unable to send request: The attempt to connect to the server failed.
Updates: Unable to access version data
```
<FULL LOG>
```

pbatard, 2023-06-10:
Not available for Windows 7.
Also, I have stopped supporting Windows 7 altogether: https://github.com/pbatard/rufus/wiki/FAQ#user-content-Windows_XP_Windows_Vista_Windows_7_NO_LONGER_SUPPORTED

closed 2023-06-10 by pbatard

oijqws, 2023-06-12:
I can't find dowload button also, the FAQ showed there should a down arrow on SELECT, but mine does not have that.
[image]

pbatard, 2023-06-12:
@oijqws: https://github.com/pbatard/rufus/wiki/FAQ#user-content-Help_I_dont_see_the_ISO_download_button

iahmtb, 2023-06-13:
it worked in one of my other devices

but why cant i download the windows 7 iso??

pineapple63, 2023-06-13:
> it worked in one of my other devices but why cant i download the windows 7 iso??
> […](#)

Microsoft removed the Windows 7 ISOs from their server