pbatard/rufus #2265: Problem with clean istall Windows 11 from pendrive made by rufus with remove all restrictions.

seba2282, 2023-06-18:
Hello, how in topic. Installation general go on old PC, but after, when is first configure devices, then after moment is bluescreen and restart PC. And it is all time. Of course as update too is the same, when isn't turned download updates and drivers by that creation update from desktop on active logged windows 10. But when we turn that option, update go success and work. Only me have that problems? Or is global?

pbatard, 2023-06-18:
> Only me have that problems?

As far as I can tell, yes, it's only you. It is not a global issue.

I'm afraid you will have to investigate that issue yourself, especially as you don't provide the log of how you created your drive.

closed 2023-06-18 by pbatard

seba2282, 2023-06-18:
How to give logs? If was putted traditional mbr ntfs for bios or uefi with csm
and only removed restrictions when made pendrive + add bonus partition for detect by old bioses that patch and rufus was downloaded from microsoft store. Earlier when I was put iso 22h2 few times ago Windows 11, wasn't that types problems. So you to fast closed topic. I thought some users if have problems too, will tell about and maybe someone tell fix. Ahhh and I forgot tell, making bootable pendrive ended succesfull without problems.

pbatard, 2023-06-18:
> How to give logs? 

Don't delete the check list that says:

> Checklist
> ---------
> - [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
> - [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
> - [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
> - [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.
> 
> Additionally (if applicable):
> - [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
> - [ ] I also tried one or more of the following:
>   - [ ] Using a different USB drive.
>   - [ ] Plugging the USB into a different port.
>   - [ ] Running Rufus on a different computer.
> - [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

It tells you how to provide a log.

> you to fast closed topic.

I tend to close topics from people who ignore the check list, because I'd rather spent my time helping people who can follow instructions.

seba2282, 2023-06-18:
Now i understand nice know.