pbatard/rufus #2267: add Estimated Time option

SaifMahmud366, 2023-06-20:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

SaifMahmud366, 2023-06-20:
showing estimated time would be a good option

pbatard, 2023-06-22:
You can get this for DD image copy by pressing <kbd>Alt</kbd>. Doing it for ISO mode is more complicated. I have some vague plans to do it, but it'll probably be years before it happens, __if__ if happens.

As such, I will close this issue as this falls in the category of:
- Things I would like to do.
- Things I am __not__ going to forget about wanting to do (i.e. I don't need a reminder)
- Things that are very low priority.

closed 2023-06-22 by pbatard