pbatard/rufus #2268: Windows User Experience window doesn't appear when using Rufus in a virtual machine (Parallels Desktop)

rakleed, 2023-06-23:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
The Windows User Experience window does not appear when using Rufus in a Windows 11 virtual machine accessed through the Parallels Desktop app on macOS on an Apple Silicon SoC computer. Otherwise, Rufus works without problems. If I do the same on a real Windows computer, then there is no such problem.

Log
---
```
Rufus Arm64 v4.1.2045 (Portable)
Windows version: Windows 11 Pro arm64 (Build 22621.1848)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
1 device found
Disk type: FIXED, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3814, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {1B0F3C05-DFD7-4BA9-B540-01AA56C5485F}
Max parts: 128, Start Offset: 17408, Usable = 31376672768 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {CE6B9EF6-22F3-49DF-92DA-4045BEBA5DD8}
  Size: 29.2 GB (31374572544 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {D4031ED1-AEF6-4076-83EA-68C8F0F5124E}
  Size: 1 MB (1048576 bytes)
  Start Sector: 61280510, Attributes: 0x9000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
Could not open ISO '\\Mac\Home\Downloads\en-us_windows_11_consumer_editions_version_22h2_updated_may_2023_x64_dvd_a494c9f3.iso': [0x00000057] The parameter is incorrect.
Could not mount Windows ISO for build number detection
ISO label: 'CCCOMA_X64FRE_EN-US_DV9'
  Size: 5.4 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: en-us_windows_11_consumer_editions_version_22h2_updated_may_2023_x64_dvd_a494c9f3.iso (5.4 GB)

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Opened \\.\PhysicalDrive1 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 29.2 GB)
● Creating UEFI:NTFS Partition (offset: 31375621120, size: 1 MB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Formatting to NTFS (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Opened \\.\PhysicalDrive1 for shared write access
Writing Master Boot Record...
Using Rufus protective MBR
Writing protective message SBR
Found volume \\?\Volume{7d9abc67-0135-48f9-9c0d-bf5aa5c37fa4}\
Disabling file indexing...
Successfully remounted \\?\Volume{7d9abc67-0135-48f9-9c0d-bf5aa5c37fa4}\ as F:
Extracting files...
Image is a UDF image
Extracting: F:\autorun.inf (128 bytes)

...[removed extracting information as otherwise GitHub won't let me publish the issue, but attached the full log as a file below]...

Extracting: F:\support\logging\winsetupetw.dll (73.5 KB)
Finalizing, please wait...
F:autorun.inf already exists - keeping it
NTFS Fixup (Checkdisk)...
Volume label is CCCOMA_X64FRE_EN-US_DV9.

Stage 1: Examining basic file system structure ...
  1280 file records processed.
File verification completed.
 Phase duration (File record verification): 26.34 milliseconds.
  0 large file records processed.
 Phase duration (Orphan file record recovery): 2.76 milliseconds.
  0 bad file records processed.
 Phase duration (Bad file record checking): 2.96 milliseconds.

Stage 2: Examining file name linkage ...
  1472 index entries processed.
Index verification completed.
 Phase duration (Index verification): 137.40 milliseconds.
 Phase duration (Orphan reconnection): 0.80 milliseconds.
 Phase duration (Orphan recovery to lost and found): 0.03 milliseconds.
  0 reparse records processed.
  0 reparse records processed.
 Phase duration (Reparse point and Object ID verification): 7.88 milliseconds.

Stage 3: Examining security descriptors ...
Security descriptor verification completed.
 Phase duration (Security descriptor verification): 11.78 milliseconds.
  96 data files processed.
 Phase duration (Data attribute verification): 2.63 milliseconds.

Windows has scanned the file system and found no problems.
No further action is required.

  30639230 KB total disk space.
   5726388 KB in 952 files.
       348 KB in 98 indexes.
     44518 KB in use by the system.
     41888 KB occupied by the log file.
  24867976 KB available on disk.
      4096 bytes in each allocation unit.
   7659807 total allocation units on disk.
   6216994 allocation units available on disk.
Total duration: 198.11 milliseconds (198 ms).
NTFS Fixup completed.

Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
1 device found
Disk type: FIXED, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3814, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {43013D4D-F788-42A8-90F1-E130B9F08D26}
Max parts: 128, Start Offset: 17408, Usable = 31376672768 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {7D9ABC67-0135-48F9-9C0D-BF5AA5C37FA4}
  Size: 29.2 GB (31374572544 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {8AD4B2DD-9CD3-47C8-98F9-AEB0A7A3B3DC}
  Size: 1 MB (1048576 bytes)
  Start Sector: 61280510, Attributes: 0x9000000000000000
```

[rufus.log](https://github.com/pbatard/rufus/files/11853849/rufus.log)

pineapple63, 2023-06-24:
> Could not open ISO '\\Mac\Home\Downloads\en-us_windows_11_consumer_editions_version_22h2_updated_may_2023_x64_dvd_a494c9f3.iso': [0x00000057] The parameter is incorrect.

it seems the problem is with Parallels and not Rufus
To me that looks like for some reason Rufus is seeing the Mac OS file system locations, and not the equivalent Windows locations

pbatard, 2023-06-24:
> To me that looks like for some reason Rufus is seeing the Mac OS file system locations, and not the equivalent Windows locations

In this case, Rufus is seeing the exact path (a [UNC path](https://learn.microsoft.com/en-us/dotnet/standard/io/file-path-formats#unc-paths)) that was provided by Windows through the standard Windows file selection dialog API.

I am guessing that, for convenience, Parallels is mapping the Mac user's `Home` directory as the Windows user directory as if it was a network drive, since that's one way to make Windows handle a non-local user directory, so you will get a UNC path.

Now, on a __real__ Windows platform, this is not an issue for WUE build number detection, which is the part that fails here (we need to detect the Windows build number of the image to know what WUE options we should provide), i.e. you can very much select a Windows ISO residing on a network drive, and therefore that uses a path like `\\nas\Home\Downloads\us_windows_11_consumer_editions_version_22h2_updated_may_2023_x64_dvd_a494c9f3.iso` and Rufus will have no problem mounting that ISO (using Windows native ISO mounting APIs, so that's not something we have much control to "fix" in Rufus) and detecting the Windows Build number.

In OP's case however, when feeding the Parallels UNC to [`OpenVirtualDisk()`](https://learn.microsoft.com/en-us/windows/win32/api/virtdisk/nf-virtdisk-openvirtualdisk) as the `Path` parameter, the function call fails with `[0x00000057] The parameter is incorrect.` (i.e. Windows did not like the `Path` that was provided) and this looks like a pure incompatibility between __Microsoft__'s `OpenVirtualDisk()` and __Apple__'s Parallels, that you are going to have to call on one of these two software companies to resolve (most likely Apple), because I am not going to write my own `OpenVirtualDisk()` just to work around an issue that has to do with running Windows on virtual hardware.

One thing you may want to try, to see if it helps, is copy the ISO away from your user directory and onto something like a `C:\ISOs` drive on the system (though I have to wonder if, in Apple aren't going to try to symlink any non-system local directory you create as a UNC to your Mac's home folder).

In short, you will need to report to Apple support that, when running parallels, Windows cannot mount ISOs that reside in the user directory when using Windows' native `OpenVirtualDisk()` API, as opposed to what is the case on real hardware, and therefore, that they need to investigate that as if breaks compatibility for applications like Rufus.

And because this issue appears to be 100% due to parallels not emulating Windows hardware as it should (so that native Windows APIs work as they would on real hardware) I will close this issue.

closed 2023-06-24 by pbatard