pbatard/rufus #2269: Feature Request: Ability to Remember Disabling "Create extended label and icon files"

theskipster00, 2023-06-27:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I find myself often forgetting to disable the "Create extended label and icon files". I have read the FAQ and understand why it is there, and don't think it should be removed, but it would be nice to have a way to have a way remember disabling it, as I never use it. Or, probably better, a setting that enables remembering it, so that one off disables don't get remembered for users who do want it on by default. This may be a bit too niche of an issue that is specific to me, but thought there may be others who similarly disable it every, or nearly every time. Cheers!

Log
---
```
N/A
```

pbatard, 2023-06-28:
That's a bit too specific to your workflow I'm afraid...

Can you detail in what manner letting Rufus create the icon and extended label impairs you?

Also please bear in mind that the idea with wanting to _nudge_ users into having a specific icon for drives that they are probably going to use for one time only (OS installation), is so that they are easily differentiated from regular drives that may contain important data.

A lot of users will right click on the drive to reformat it after usage, and, as such, I'd rather inconvenience a few users like you by having the default to creating the icon always, than end up with __very__ irate users, who happened to format the wrong drive by mistake because it didn't have an easily recognizable icon that told them it was safe to select to reformatting...

theskipster00, 2023-06-28:
Thought that may be the case, all your points make sense. Thanks for the insight :)

closed 2023-06-28 by theskipster00