pbatard/rufus #2270: Error: Undetermined error while formatting making WinToGo WIn11Pro from WIn10Prox64

Bub2480, 2023-06-28:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
System is Win10Pro x64 22H2.  
Attempting to create a Win11Pro WinToGo USB using Sandisk Extreme PRO SDCZ880 512gb in a USB3 port
Downloaded Win11 ISO directly from Microsoft 6-26-2023.
I've tried zeroing the drive from within Rufus, as well as using Paragon Hard DIsk Manager.  
I've tried ALT- but it didn't work either.  
I've tried checking for bad blocks, none found.  
I've tried a different USB3 port.  No change.
I've tried using a Sandisk Ultra USB 3.0 64gb USB SUCCESSFULLY using the same ISO and Rufus settings. 
My reason for using the Extreme PRO is speed.  It is a SSD in a USB stick, verses just a flash drive.  Much faster.
All the other issues were bad blocks or issues running Rufus on WIn11 not being able to format the USB, unless I missed one.  I didn't go back farther than 2018 as I am running 22H2 WIn10Pro.  

Log
---
```
<FULL LOG>
Rufus x64 v4.1.2045 (Portable)
Windows version: Windows 10 Pro x64 (Build 19045.3086)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found UAS (USB 3.0) device 'SanDisk 3.2 Gen 1 UAS Device' (0781:5588)
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x87
Disk type: FIXED, Disk size: 512 GB, Sector size: 512 bytes
Cylinders: 60083, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 0
Disk GUID: {519EAA18-8F12-4ADF-AC7F-C860527D9872}
Max parts: 128, Start Offset: 17408, Usable = 494206417408 bytes
Scanning image...
ISO analysis:
[rufus.log](https://github.com/pbatard/rufus/files/11898558/rufus.log)

  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'ESD-ISO'
  Detected: Windows 11 ISO (Build 22621.1702)
  Size: 4.5 GB (Projected)
  Note: File on disk is larger than reported ISO size by 586 KB...
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.esd (version 0.14.0+)
Using image: Windows.iso (4.5 GB)
Checking for Rufus updates...
Checking release channel...
No new release version found.
WIM extraction method(s) supported: wimgapi.dll
[... 180 lines cut ...]
  Size: 260 MB (272629760 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: Microsoft System Reserved Partition
  Name: 'Microsoft Reserved Partition'
  Detected File System: (Unrecognized)
  ID: {E4048907-291A-42B5-826A-AF5B981C74AB}
  Size: 128 MB (134217728 bytes)
  Start Sector: 534528, Attributes: 0x0000000000000000
Partition 3:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: (Unrecognized)
  ID: {B1741AE6-4CD6-491E-95A2-440B6637144C}
  Size: 459.9 GB (493798538752 bytes)
  Start Sector: 796672, Attributes: 0x0000000000000000

```
[image]
[image]
[image]
[rufus - Copy (2).log](https://github.com/pbatard/rufus/files/11898240/rufus.-.Copy.2.log)
[rufus - Copy (3).log](https://github.com/pbatard/rufus/files/11898241/rufus.-.Copy.3.log)
[rufus - Copy.log](https://github.com/pbatard/rufus/files/11898242/rufus.-.Copy.log)
[rufus.log](https://github.com/pbatard/rufus/files/11898560/rufus.log)

pbatard, 2023-06-28:
Please try from a different computer.

Considering that, with the number of times Rufus is being downloaded each month, you are unlikely to be the only person to have ever tried to use a SanDisk 3.2 Gen 1 UAS Device 512 GB with Rufus, and that nobody else appears to report a similar issue, I first have to assume that the issue has to do with your Windows environment. Therefore it is crucial that you try on multiple Windows machine (preferably with a Windows 11 machine thrown into the mix if you can) to confirm that it's a global issue, because the only way I may ever remotely look into it is if it can be validated that the issue you report happens on all modern versions of Windows with this specific drive.

But even then, with SanDisk finding nothing better than selling completely different hardware under the same model name (because I do happen to have a "Sandisk Extreme PRO SDCZ880" (128GB version) that also identifies as `0781:5588`, but as opposed to yours, it's not FIXED or UAS and, obviously, I don't see your issue), I'd say there's only a 20-30% chance that, even if I try to purchase the exact same model as yours, I will end up with the actual same hardware, and as you can understand, I'm not going to be that thrilled about wasting more than €100 on a drive, if it's going to be useless to replicate an issue.

Which means that, at this stage, I'm going to need two things to happen before I consider looking into it further:

1. You, confirming that you can replicate the same issue on at least 2 other Windows computers. And yeah, I understand that it might be troublesome, but we have no option but to filter out a purely environmental issue first and foremost. And, in case this needs to be pointed out, please don't be tempted to misreport the outcome, because we will find out.
2. At least 1 or 2 separate persons reporting a similar issue.

For now, what I can advise you to try is switch to using VDS for formatting by pressing <kbd>Alt</kbd>-<kbd>V</kbd> in Rufus. Then you should see the line `Formatting to NTFS (using VDS)` in the log (instead of `(using IFS)`), and see if that helps.

pbatard, 2023-07-04:
I will close this issue for now, as I am seeing no evidence that this isn't an environmental issue with the system OP tested with.

Again, this does not mean that I am dismissing the issue, or that I will not look into it further __if__ someone can confirm that they experience the same issue on a different computer than the one OP used. But it doesn't make sense to leave this issue open until someone does.

closed 2023-07-04 by pbatard

arandomusertoo, 2023-08-04:
Happens to me on another computer (one that I've successfully used in the past with same exact settings).

`Rufus x64 v4.2.2074
Windows version: Windows 10 Enterprise x64 (Build 19045.3086)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found UAS (USB 3.0) device ' USB DISK 3.2 UAS Device' (13FE:6500)
1 device found
Disk type: FIXED, Disk size: 512 GB, Sector size: 512 bytes
Cylinders: 60801, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x78D50683
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: exFAT (0x07)
  Detected File System: exFAT
  Size: 465.8 GB (500103667712 bytes)
  Start Sector: 8192, Boot: No
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CPBA_X64FRE_EN-US_DV9'
  Detected: Windows 11 ISO (Build 22621.1992)
  Size: 5.4 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: en-us_windows_11_business_editions_version_22h2_updated_july_2023_x64_dvd_a031f913.iso (5.4 GB)
Checking for Rufus updates...
Checking release channel...
No new release version found.
WIM extraction method(s) supported: 7-Zip, wimgapi.dll
WIM apply method supported: wimgapi.dll
Opening: \\.\CDROM0/sources/install.wim:[0] (API)
Extracting: C:\Users\Me\AppData\Local\Temp\Ruf7774.tmp (From [1].xml)
Closing: \\.\CDROM0/sources/install.wim
Will use 'Windows 11 Enterprise' (Build: 22621, Index 3) for Windows To Go
Selected Windows User Experience options:
• Bypass online account requirement
• Disable data collection
• Use 'Me' for local account name

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Opened \\.\PhysicalDrive3 for shared write access
Analyzing existing boot records...
Drive has a Windows 2000/XP/2003 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating EFI System Partition (offset: 1048576, size: 260 MB)
● Creating Microsoft Reserved Partition (offset: 273678336, size: 128 MB)
● Creating Main Data Partition (offset: 407896064, size: 465.4 GB)
Waiting for logical drive to reappear...
Formatting to NTFS (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Volume is in use
[0x00000001] Incorrect function. - Retrying...
[... 163 lines cut ...]
Partition type: GPT, NB Partitions: 3
Disk GUID: {A154C46B-6F5C-45DA-A243-8C7897B7746D}
Max parts: 128, Start Offset: 17408, Usable = 500107827712 bytes
Partition 1:
  Type: EFI System Partition
  Name: 'EFI System Partition'
  Detected File System: (Unrecognized)
  ID: {6CA0CEFD-1E51-495F-8A89-CE89341FDB94}
  Size: 260 MB (272629760 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: Microsoft System Reserved Partition
  Name: 'Microsoft Reserved Partition'
  Detected File System: (Unrecognized)
  ID: {CE608E6A-CA67-46A7-A17F-9583E03D66CC}
  Size: 128 MB (134217728 bytes)
  Start Sector: 534528, Attributes: 0x0000000000000000
Partition 3:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: (Unrecognized)
  ID: {997F47F6-CA56-4A8B-B9B1-B87849212257}
  Size: 465.4 GB (499699945472 bytes)
  Start Sector: 796672, Attributes: 0x0000000000000000
`