pbatard/rufus #2272: Problem when booting the program

Tarq2, 2023-07-01:
When trying to burn, the program disappears and does not complete the process rufus-4.1

pbatard, 2023-07-01:
Install [DebugView as was requested from the log that you decided to ignore and provide a log.

Yes, even if the application disappears or crashes, you __will__ get a log from DebugView.

Unless you do provide a log withing 24 hours, I will simply close this issue as I don't have __ANY__ details that allow me to even remotely start to help you.

Tarq2, 2023-07-01:
> 

[DESKTOP-4N1770N.LOG](https://github.com/pbatard/rufus/files/11925590/DESKTOP-4N1770N.LOG)

pbatard, 2023-07-01:
I need to know the ISO you used, and where exactly you downloaded it. It appears that the application crashes when parsing the ISO, which means that, unless I can get access to that specific ISO, I'm not going to be able to do much...

Tarq2, 2023-07-01:
> I need to know the ISO you used, and where exactly you downloaded it. It appears that the application crashes when parsing the ISO, which means that, unless I can get access to that specific ISO, I'm not going to be able to do much...

I downloaded ios via rufus

pbatard, 2023-07-01:
> I downloaded ios via rufus

It still doesn't tell me what ISO it was. The ISO has a name, architecture, language, etc. What are they?
And do you actually observe a crash with an error, or does the application just go away.

Please bear in mind that I am not in front of your computer, I don't have access to your data and I can't read your mind, so you have to provide enough details for us to try to help you, as I still don't have enough data to do anything about your issue here.

JonnyTech, 2023-07-01:
`Windows version: Windows 10 Pro x86 (Build 19045.3086)`

Is this correct, a 32 bit version of Windows?

pbatard, 2023-07-01:
No, I am asking about the version of Windows you downloaded through Rufus. `Windows version: Windows 10 Pro x86 (Build 19045.3086)`  is the version of Windows you are running on.

When you downloaded the ISO from Rufus, as you state you did, you selected:
- A version of Windows (Windows 11, Windows 10, Windows 8)
- A release of Windows (most likely 22H2)
- A language
- An architecture

And then after the download completed, you ended up with a `.iso` file in the folder where you chose to download it, and that `.iso` file has a name, such as `Win11_22H2_EnglishInternational_x64v1.iso`.

I need to know all of these details.

Tarq2, 2023-07-01:
> And do you actually observe a crash with an error, or does the application just go away.

I did not notice an error that occurs, that the application disappears at the beginning of burning

Tarq2, 2023-07-01:
> It still doesn't tell me what ISO it was. The ISO has a name, architecture, language, etc. What are they?

Win10_22H2_Arabic_x32v1

pbatard, 2023-07-01:
> I did not notice an error that occurs, that the application disappears at the beginning of burning

So you can press `START` after you selected the ISO?

In that case the log you provided with DebugView is incomplete because there should be additional lines after:
```
00000078	17.56792259	[6040]   Image does not have a Boot Marker 	
```

So, can you press `START` after you select the ISO, and do you get any additional lines in DebugView after `Image does not have a Boot Marker`?

JonnyTech, 2023-07-01:
[OT] @pbatard sorry, I was just checking on the version of Windows that was **running** Rufus because I recall someone else that I know having a similar problem with a 32 bit Windows 10, they never provided me their logs (else I would have submitted an issue here) but the ISO worked correctly on a 64 bit Windows 10. They were in the UAE so would have been an Arabic ISO too, if that helps with this issue.

Tarq2, 2023-07-01:
> So you can press `START` after you selected the ISO?

no

pbatard, 2023-07-01:
@JonnyTech, I missed that you weren't OP when replying, which didn't really help with trying to get the data needed. Windows 10, which is what OP uses, __does have__ 32-bit versions. It's only Windows 11 that dropped x86_32 support.

Tarq2, 2023-07-01:
> So, can you press `START` after you select the ISO, and do you get any additional lines in DebugView after `Image does not have a Boot Marker`?
[DESKTOP-4N1770N.LOG](https://github.com/pbatard/rufus/files/11925653/DESKTOP-4N1770N.LOG)

I tried again and this reading of the problem and error

pbatard, 2023-07-01:
@Tarq2, thanks, I can finally replicate your issue!

It only seems to happen when using the _x86 version of Rufus (64-bit is fine), and I can also see it when scanning other Windows ISOs besides `Win10_22H2_Arabic_x32v1.iso`.

This also appears to be an issue that's restricted to the MinGW built version and not the MSVC built version, because as far as I can see, the MSVC built x86 version does not crash (because if it did, I would have seen reports from Windows Store users, who use an MSVC build).

@Tarq2, can you please try installing the official Rufus App from https://apps.microsoft.com/store/detail/rufus/9PC3H3V7Q9CH and see if that works better for you?

Now that I finally have enough data, I will of course try to figure out why the x86 MinGW crashes while parsing these Windows ISOs...

pbatard, 2023-07-01:
The issue appears to have to do with the delay loading of the virtdisk library which we enforce to prevent hijacking of the application through DLL side loading (See #1877).

From what I can tell, __for MinGW32 only__ (i.e. MinGW64 does not have this issue) we run into the same problem as the one I described [here](https://github.com/pbatard/rufus/issues/1877#issuecomment-1109683039), with the MinGW delay loading code producing a literal _"That's it, I'm outta here"_ from the runtime when it encounters an issue.

More specifically, after a Windows ISO is scanned, we try to detect the Windows build from that ISO by mounting, __using the Windows virtdisk APIS__ it so we can poke at the relevant files. And it looks like the _delay load_ [call to `OpenVirtualDisk()` from MountISO()](https://github.com/pbatard/rufus/blob/c717c7789b4f79a4cb500f9287f5d355fed591a4/src/iso.c#L1726) crashes the whole app when it's the x86 MinGW32 compiled version.

And, no, this is not an overall issue with delay loading for MinGW32, as we have other delay loaded calls in the application, and especially one that validates the ISO download script before we launch it, and those ones do not produce a crash.

<Sigh> I guess it's back to direct hooking into DLL API calls, like I had to do for `wininet`, since neither Microsoft nor MinGW are of any actual help when it comes to preventing side-loading vulnerabilities...

JonnyTech, 2023-07-01:
@pbatard just out of curiosity, is it only foreign language ISO's or Windows hosts? I contacted the others that I mentioned earlier and they claim to have no issue with English ISO's and English 32 bit Windows. Yet they still won't provide a log!

pineapple63, 2023-07-01:
@JonnyTech i cant say if english US ISOs are affected (as I don’t want to download ISOs that aren’t of any use to me), but i tested an English International 32 bit ISO on one of my systems, and it caused Rufus to crash (when using the 32 bit version of Rufus)

pbatard, 2023-07-01:
It's all the recent Windows ISOs that I've tested, English, other languages, x86/x64, whatever, which makes sense since it's a crash as soon as a specific function call is invoked, and that function call will be invoked for any modern Windows ISO.

Please be very mindful that:
- This will not happen with any x64 version of Rufus
- This will not happen with the x86 version from the Store either

The only version where you will see that is if you use the `rufus-4.1_x86.exe`  version from https://rufus.ie since this issue was introduced in the second commit [0df68b6af980b3cdbb128f46ca6faa1704db9581] we added after the 4.0 release, where we switched to delay loading the virtdisk library.

Tarq2, 2023-07-02:
I downloaded ios X64, unlike I previously installed ios X86, and it booted successfully. Please wait for the installation to complete and then come back and report that the issue has been resolved

closed 2023-07-02 by pbatard

Tarq2, 2023-07-06:
The problem exists

Tarq2, 2023-07-06:
What is the version of Rufus that fits my system?

commit f9370e00 (2023-07-02): [iso] fix a crash when parsing Windows ISOs with MinGW x86_32

* MinGW32's delay loading functionality is not yet up to par with MSVC's and especially, for
  some libraries (wininet, virtdisk) attempting to delay load them simply crashes the runtime.
* This results in the MinGW32 version of the app crashing when selecting a Windows ISO, as we
  will then try to mount the ISO using virtdisk to poke the build version. Note that this crash
  does not happen with the MinGW64 version or with MSVC.
* Closes #2272.
* Also fix a Coverity warning in SaveImageThread() and improve the VHD saving code.