pbatard/rufus #2273: the latest mingw from Actions was flagged as wacatac virus my defender lol

infr-automation, 2023-07-02:
Hi Paul, sorry to 'waste time of senior dev' however just making one simple note that the latest build from actions mingw was flagged as Trojan:Script/Wacatac.H!ml virus.

I think there is a button to submit for review from my end however maybe there is also something you can do if you have some communication with MS.

pbatard, 2023-07-02:
https://github.com/pbatard/rufus/wiki/FAQ#user-content-Antivirus_X_reports_that_Rufus_contains_malware

I genuinely no longer have time for this shit.

closed 2023-07-02 by pbatard

infr-automation, 2023-07-04:
After a lot of information gathering, I have come to the conclusion that it is a good idea to automate the submission through API which seems to be the MS Security Graph or something like that (wow they keep all that data that's kinda scary lol I hope they can manage it correctly)  during the CI CD pipeline. As part of DevSecOps good practices.

And another interesting AV vendor to communicate with would be Eset according to the information I have so far.

So let's say that corporations do grand things and sometimes it can get out of control such as create grief for the users, limit the users, offend the users, and spy on the users. And history have shown that corporations are not to be trusted and are always  to be tripple checked.

So the way to do that would be to personally know the guy, who makes the algorithms for that beast mentioned above.