pbatard/rufus #2276: Rufus.ie website is currently down.

Yomo42, 2023-07-07:
Title. There didn't seem to be any other channel to make this known.

Hrxn, 2023-07-07:
Not down for me atm.

daondev, 2023-07-07:
Website downed for me too, it seems like DNS servers supporting rufus.ie domain has been downed.
Acquired previous IP information by searching on google, and set IP by force on 'hosts' file of my Windows PC.
And download of additional file. Website also accessible.
Won't write the IP info for security reason...

brianwp3000, 2023-07-07:
For anyone else unable to use Rufus because it fails to download syslinux files, an alternative workaround is to download the files directly from the [rufus-web](https://github.com/pbatard/rufus-web/tree/gh-pages/files) repository. Then follow the instructions in the [Usage Notes](https://github.com/pbatard/rufus/wiki/Usage-Notes#where-to-manually-copy-the-syslinuxgrub-files-if-rufus-cant-download-them) for placing those files on your local machine.

BRAINIFII, 2023-07-07:
For now you can add host entries https://securitytrails.com/domain/rufus.ie/dns get the IP from here.

It will also download the grub files in rufus

host file:
<ip-address>   rufus.ie

> Website downed for me too, it seems like DNS servers supporting rufus.ie domain has been downed. Acquired previous IP information by searching on google, and set IP by force on 'hosts' file of my [...]

BRAINIFII, 2023-07-07:
For now you can add host entries https://securitytrails.com/domain/rufus.ie/dns get the IP from here.

It will also download the grub files in rufus

host file:
(ip-address)   rufus.ie

> Website downed for me too, it seems like DNS servers supporting rufus.ie domain has been downed. Acquired previous IP information by searching on google, and set IP by force on 'hosts' file of my [...]

Sislik42, 2023-07-07:
I have the same issue. I even tried to ask 1.1.1.1 and it also can't resolve the hostname...

pbatard, 2023-07-07:
Sadly I can confirm the issue, and it appears to be an problem with the name servers from my registrar, which somehow dropped all resolutions for `rufus.ie` as well as `akeo.ie` (since I have not changed anything from my DNS records in months, so it's not a misconfiguration from my side).

Obviously, I have created a support ticket with them and am waiting for them to fix the issue.

Basically, you can see that the issue if on their end if you issue `nslookup` with the IP of their `ns1.blacknightdns.com` name server (which is one of the default name servers I am using):
```
# nslookup rufus.ie 78.153.203.12
Server:         78.153.203.12
Address:        78.153.203.12#53

** server can't find rufus.ie: REFUSED
```

BRAINIFII, 2023-07-07:
Hey @pbatard why don't you shift your ns to cloudflare like it my just help with many other things if you might like

pbatard, 2023-07-07:
I don't think that can work unless the name servers from the registrar can tell CloudFlare the records to which the DNS queries should resolve first... which it won't be able to do until the registrar fixes this issue.

Because, if if it was as easy to tell CloudFlare that whatever domain should resolve to whatever record, anybody could pretend that they own microsoft.com (or github.com) and make the CloudFlare name server resolve these domains to malicious ones.

So, there needs to be validation of ownership occurring, and, if it's like any other validation of domain ownership that I've see (from Google for custom gmail domains to GitHub for custom pages domains), this is being enacted by adding special entries to the DNS records served by the registrar, whereas  the issue here is that the registrar is not serving anything from my domains...

Oh, and it definitely looks like a general issue with this specific registrar, as libusbx.info, which I don't own, but which I know is also registered with the same registrar, also no longer resolves as of today.

Someone, somewhere, must have a heck of a dreadful friday night trying to urgently restore a production database, that has just been deleted by some intern or something

BRAINIFII, 2023-07-08:
Oops, my mistake! I didn't read your message carefully having problem with the NS itself, OK THATS BAD..!! hope it gets fixed quickly..

pbatard, 2023-07-08:
The amount of good will I have for BlackNight as a resgistrar is starting to wear __very__ thin, as I am still waiting for them to get by to me with some progress on this issue (I had a reply from a support person telling that they would look into and get by to me by 3 hours ago... but that hasn't happened).

If they still can't get their shit together within the next 8 hours, I'm going to take my business to a registrar that both doesn't screw up their name servers on a whim and also actually follows up with customers when issues happen, because there's really only so much patience with having my domains down for 48 hours, because of somebody else's screwup, that I can have...

pbatard, 2023-07-08:
Site should be accessible again, though I am still waiting for a complete resolution and explanation of the issue.

JonnyTech, 2023-07-08:
> Site should be accessible again

Confirmed

po-karlsson, 2023-07-08:
I still can’t reach rufus.ie...

pbatard, 2023-07-08:
> I still can’t reach rufus.ie...

You may have to wait for your name servers to refresh. It can take up to 48 hours.

po-karlsson, 2023-07-08:
Ok, thanks!

Jyun5978, 2023-07-08:
[image]
It works now!

pbatard, 2023-07-09:
Since the website is accessible again, I'm going to close this issue now.

closed 2023-07-09 by pbatard