pbatard/rufus #2280: Is Rufus support USB Type-A (3.0 and greater) and USB Type-C?

GauravSahare1993, 2023-07-14:
I have SanDisk Ultra USB Pen Drive (64 GB) that supports USB 3.0, I connected my Pen Drive to my desktop computer for making USB bootable. Successfully works with USB 2.0 port but I did not test on USB 3.0 (to make bootable).
So my question "Is Rufus support USB Type-A (3.0 and greater) and USB-C?
I never tested, because I am not sure that bootable USB works proper (after connect to USB 3.0 port). I tried to find related information on official site and even on Github, but no information available regarding from this topic.

pineapple63, 2023-07-14:
Rufus does support USB 3.0 and USB C devices

pbatard, 2023-07-14:
Yes, the type of port doesn't matter.