pbatard/rufus #2282: OpenDOS compatible with fat32

4nd3r50ncr, 2023-07-15:
Wouldn't it be possible to make OpenDOS available on rufus to install the same as FreeDOS and MS-DOS? It would be interesting to have these possibilities available, it would be more complete, sys.com can make the base installation of OpenDOS work, I already tested it a while ago, it just needs to be enabled in the source code before compilation, and it works formatted in fat32 by Windows. Beautiful working! But it would be more beautiful if it was natively with rufus, msdos 6 would be interesting too...

pbatard, 2023-07-15:
> Wouldn't it be possible to make OpenDOS available on rufus to install the same as FreeDOS and MS-DOS?

I have no plans to do so. I'm afraid that I am far from sharing your view of how this would actually benefit Rufus users. As a Free Software developer, I'd much rather that people who think FreeDOS is lacking features compared to OpenDOS/DR-DOS do make an effort to contribute to the __still actively maintained__ FreeDOS to bring them there.

The cost/benefit ratio of adding and maintaining OpenDOS is simply too high.

>  msdos 6 would be interesting too...

MS-DOS 6 has been asked over and over, but I can not provide it in Rufus because it is copyrighted by Microsoft, and redistributing it for installation by Rufus would be copyright infringement. The only reason MS-DOS (WinME version) was reinstated in Rufus recently (after being removed for many years precisely because we would have committed copyright infringement otherwise) is because we found a way to download a DLL that includes the full WinME MS-DOS floppy image directly from Microsoft (and therefore, since it is not distributed by Rufus or included in Rufus, but comes from Microsoft, it becomes legal to provide it).

So, unless you can find a complete copy of MS-DOS 6 __publicly__ accessible on Microsoft's web servers, MS-DOS 6 will never be provided by Rufus.

Since I believe I have answered your question, I will close this issue,

closed 2023-07-15 by pbatard

4nd3r50ncr, 2023-07-15:
Okay, from an MS-DOS point of view, but OpenDOS is available, still, it wouldn't cost anything to keep OpenDOS working on rufus, OpenDOS and FreeDOS have certain similarities with compatibility, so the fact that it works on fat32 is very important. good, and it's free. It would be one more option, I do all this manually, to run opendos through my usbstick using FreeDOS, then I managed to compile OpenDOS, it's no use explaining if you're not interested, it was magical to see it work, and the joy of having another one available for use however you want.

pbatard, 2023-07-15:
>  it wouldn't cost anything to keep OpenDOS working on rufus

Please don't assume that, because Rufus is free, my time is also free or that adding __and maintaining__ a feature doesn't have a cost.

Please read https://github.com/pbatard/rufus/wiki/FAQ#user-content-Why_cant_you_just_add_the_feature_I_want and understand that everything looks like a 5 minutes job... when you're not the one who has to do it.

4nd3r50ncr, 2023-07-15:
Thank you for your kind reply.