pbatard/rufus #2283: Simple Question not an issue feel free to close it when you reply

willzyx-hub, 2023-07-16:
Hi Bro, i wanna to ask how did you get Verified Publisher when your cert doesn't even Installed on the Windows itself ?

pbatard, 2023-07-16:
I purchased a very expensive [EV code signing certificate](https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/code-signing-cert-manage#get-or-renew-a-code-signing-certificate) since that's the certificate you need to be able to get UEFI bootloaders signed for Secure Boot.

And, sadly, as is the case with almost anything, the more you look like you're wealthy, the more people and companies say they will trust you...

closed 2023-07-16 by pbatard

willzyx-hub, 2023-07-17:
> I purchased a very expensive [EV code signing certificate](https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/code-signing-cert-manage#get-or-renew-a-code-signing-certificate) [...]

Btw if there is any Free Alternative, to make the .exe and Bootloader signed ? I signed with my Free Digicert certificate but it only get Verified Publisher only if the cert is Installed on the system.