pbatard/rufus #2287: Suggestion of creating iso Images

modz2014, 2023-07-22:
Hi I was wondering if you could have the create iso option once disabling the tpm checks etc so the iso can be used in VirtualBox etc others I have to burn the iso to USB first then take the files off the USB and then create another iso file to be used with VirtualBox

pbatard, 2023-07-22:
Not planning to.

You can already create an uncompressed VHD (which for all intent and purposes can be be used as a DD image in Virtual Box), and as you said yourself, if you really want to use an ISO, you can easily create it yourself.

See https://github.com/pbatard/rufus/wiki/FAQ#user-content-Creating_and_using_a_VHD_from_scratch.

And, in case you are unhappy with this answer, please make sure you read https://github.com/pbatard/rufus/wiki/FAQ#user-content-Why_cant_you_just_add_the_feature_I_want.

closed 2023-07-22 by pbatard

modz2014, 2023-07-23:
seems pointless using other software when refus cn already do the stuff already when burning it to a usb so adding the feature to refus would be good just saves a lot of time ect just doing what you stated in the first link is complicated for end user if they dont what to do but yer i guess its up to you