pbatard/rufus #2290: Since 4.1 the file picker/explorer does not show any files or folders

natrius, 2023-07-25:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I opened rufus-4.1exe and after pressing "Choose file" i get presented with nothing to choose, even if i change to "Show all files".
[image]

rufus-4.0.exe (same folder)
[image]

Log
---
```
Rufus x64 v4.1.2045
Windows version: Windows 10 Enterprise x64 (Build 19045.3208)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0407 (de-DE)
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
0 devices found
```

I mean, i'm just using the 4.0 now, but with 4.1 its still possible to simply drag&drop the iso into rufus, so its not too bad. But still not good. 

Best regards

pbatard, 2023-07-25:
Thanks for the report but I'm afraid I am unable to replicate your issue, which means that it is next to impossible to investigate and potentially fix it (since I can't simply reverse changes that happened between 4.0 and 4.1 at random and "hope" that one of these will fix your issue).

Especially I can __not__ replicate your problem on a freshly installed, fully up to date version of Windows 10 Enterprise German (Build 19045.3208):
[image]

Note that I am obviously using the Eval version (downloaded from https://www.microsoft.com/en-us/evalcenter/download-windows-10-enterprise), but, for all intents and purposes, the Eval version should behave exactly the same as the regular version, as the only difference between the two is that Eval will expire after 90 days. And I obviously made sure to pick the German version, which, as you can see from the screenshot, I also installed with the default German locale.

Therefore, and especially as nobody else is reporting this issue, there is clearly an environmental component to your problem, that has to do with the manner in which you (or your company) configured your Windows 10 edition.

I could for instance see a corporate security solution that hijacks the file open/save dialogs from Windows to filter out what people can and cannot access according to the application being run (for instance, Rufus 4.0 might be whitelisted for access, but Rufus 4.1 might not).

At this stage, unless you can install a clean Windows 10 Enterprise edition and tell me how one should configure it so that it can replicate your issue, there is unfortunately very little I can do on my side because, again, I can not attempt to fix issue by reverting changes at random.

I would therefore encourage you to either find out what setting of your environment is tied to the issue, or, install Visual Studio 2022 on your problematic Windows 10 environment, and then clone the git Rufus source code and revert commits between 4.1 and 4.0 until you find the one that appears to trigger the issue (and yes, it would logically make sense that it's 0df68b6af980b3cdbb128f46ca6faa1704db9581, however, I see nothing in that commit that can explain the issue you describe, as most of it is dead code removal that should logically have no impact on the file dialog).

I would also appreciate if you could test with Rufus 4.2 BETA (downloadable from https://rufus.ie) and with the Microsoft Store version of Rufus (installable from https://apps.microsoft.com/store/detail/rufus/9PC3H3V7Q9CH or from the Microsoft Store) to see if those versions also exhibit the same issue.

natrius, 2023-07-26:
I appreciate your lengthy explanation and answer! I will use that for myself as an good example when trying to answer questions. 

The other computer i tried it on was should have the same Windows version as mine so i was already afraid it just affects me - especially after i did not find another issue about it. 

I downloaded the Rufus 4.2 BETA and i have the same problem, as well as with the download through the windows store (first time i used the windows store :D ) with the same issue. I forgot to try it at home on my personal computer, i'm afraid. 

I don't think i'm capable to clone and revert the repo and the commits, i might try it later in time but don't count on it. 

Thank you for you amazing program and your good answer, as i have a workaround (drag&drop into the rufus window) it is fine for me to close the issue if you like. If i get a new computer i will try it again.

pbatard, 2023-07-26:
Thanks for understanding.

I still hope you will have a chance to explore this further, as, if you are experiencing this, I expect other people to experience it too, and I sure wouldn't mind having an explanation for what's going on, and how that ties into a code alteration we applied between 4.0 and 4.1. There's probably some specific Windows component configuration that can trigger this, that may very well happen on all Windows version (rather than just Enterprise, or a specific localization).

As one last ditch effort, can I ask you to please download an test with https://files.akeo.ie/rufus/rufus-4.2_TEST.exe which is a version that reverts [the only code change between 4.0 and 4.1 that I can remotely envision as having an impact](https://github.com/pbatard/rufus/commit/0df68b6af980b3cdbb128f46ca6faa1704db9581#diff-98c3b5a3b66482fd9da0cc9f0ef6e3d8a5f50cfe25a024eb5ba8bdf3f159dd11) and see if that appears to fix your issue?

If not, then I guess I will have to wait until somebody else reports a similar problem to see if they can investigate it...

pbatard, 2023-07-30:
@natrius, any update on the above?

I'd really like to know if the TEST version of Rufus fixes your issue or not. So if you could let me know, I'd appreciate it.

natrius, 2023-07-31:
Sorry for the long wait, i tried it and unfortunately no change on my laptop, still no file found. (unfortunately github does not want screenshots right now)

[image]

pbatard, 2023-07-31:
Thanks for the test. For the record, I would have believed you even if you didn't attach a screenshot.

Since that didn't work, let me throw one last test version for you to try, that you can download at https://files.akeo.ie/rufus/rufus-4.2_TEST2.exe. This reverts the only other change that I can remotely see as having an impact to the file selection dialog.

If that still doesn't fix it, then I really have no idea what change might have intervened between 4.0 and 4.1 to have a play with this issue, and I'm going to have to assume that your problem is due to explicit filtering by a security solution on your system or something, because it doesn't really make sense otherwise.

natrius, 2023-08-03:
I tried it right now and unfortunately no change, still no files found in the windows :/ I'm  not sure when i get a new laptop issued, but i will try again then. But others are using rufus as well and do not report the problem. Not sure about their laptops, but the windows version is definitely the same. 

Many thanks for the infos and trying!

pbatard, 2023-08-03:
OK, thanks a lot for the tests.

I'm afraid that, at this stage, it's going to take someone that experiences the issue to install Visual Studio 2022 and clone the Rufus source from the this repository, and then revert each commit between 4.1 and 4.0 to give us a clue if a specific change from the codebase is at play, and if so, what commit introduced it.

I'm afraid that, since there's not much of anything more I can do to investigate this issue, that clearly also has an environmental component (else a lot of people would be reporting the same in this tracker), I'm going to have to close it for the time being. But I will make sure to direct people who experience a similar problem to it, and if there are new development that allow me to investigate further (such as someone having cloned the sourcecode and pinpointed a specific commit that triggers the issue in Visual Studio), I will reopen it.

closed 2023-08-03 by pbatard