pbatard/rufus #2292: Feature Request: Give option to download the portable version.

as-muncher, 2023-07-26:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Well, I just noticed that, using Firefox, if I press the Insert Key, and type an "x" in the above checkboxes, the x does not overwrite the existing space that is there.  It's not a Windows issue, but a Firefox one.

But, I use the portable version of rufus all the time, and create a shortcut for it, which I then post on the taskbar.  But when there's a rufus update, I only get the option to download the non-portable version.  So why not create a checkbox for the non-portable version, and another line with another checkbox, if I want to download the portable version?  That way, I don't have to open my web browser and download rufus off the web page.

Log
---
```
<FULL LOG>
```

JonnyTech, 2023-07-27:
Please read the FAQ:

https://github.com/pbatard/rufus/wiki/FAQ#user-content-How_comes_the_portable_and_regular_versions_are_binary_identical

The portable and non-portable versions of Rufus are identical. Add a "p" to the filename and Rufus will start in portable mode. Example: `rufus.exe` to `rufusp.exe`.

as-muncher, 2023-07-27:
I did read the FAQ and saw that they were identical, but did not know that you could edit the command line and add a "p" to download the portable version.  That being said, I see on your website that there are 4 versions available, so it might be something to consider to have those options available to checkmark for downloading.

as-muncher, 2023-07-27:
And actually, just to let you know, I am not able to add a "p" in the command line to download the portable version, using Rufus 4.1p.

as-muncher, 2023-07-27:
So this also means, that I cannot download 4.2 build 2075, because only 4.2 build 2074 is available on the website.

pineapple63, 2023-07-27:
> So this also means, that I cannot download 4.2 build 2075, because only 4.2 build 2074 is available on the website.

I don’t think there is a build 2075 (4.2 is 2074)

pbatard, 2023-07-27:
This was a typo in the version file. There is __no__ build 2075 but [I mistyped the build number in the update file](https://github.com/pbatard/rufus-web/commit/6185905984601591460a0504a100c9411dc3bd7c) so the internal updater tries to download one. I have now fixed this (though it may take a few hours for the changes to propagate).

closed 2023-07-27 by pbatard

as-muncher, 2023-07-28:
@pbatard I don't know why you closed this as completed, because it's not completed.  For one thing, did you fix the command line so that I can add a "p" and thus download the portable version?  Or was I supposed to download the regular version but then affix a "p" to it?  Also, what about downloading the other versions, such as the 32 bit version or the arm64 version?

pbatard, 2023-07-28:
>  Or was I supposed to download the regular version but then affix a "p" to it?

Yes. Or you can just download the regular version and use it in the same directory as where you have a `rufus.ini` file, because then Rufus will start in "portable mode".

> Also, what about downloading the other versions, such as the 32 bit version or the arm64 version?

They are __clearly__ available for download on the homepage at https://rufus.ie.

as-muncher, 2023-07-30:
@pbatard, Well, the whole point of me raising this issue was that, if I was using the installable version of rufus, then the update would download that version, or if I'm using the portable version, then that version would be downloaded.  I'm not even sure why you have 2 versions - you might as well just make every version have a "p" at the end.  The point was to not have to go to the web and download it off the page...

pbatard, 2023-07-30:
If you downloaded and ran the portable version then Rufus will download the "portable" version because then you will have a `rufus.ini` file and, regardless of whether there's a 'p' in the name or not, Rufus will detect that and run in portable mode.

So there is absolutely no point in downloading the "portable" version, because you get the __exact same__ outcome when downloading the "regular" version (that knows whether it should act as portable or regular depending on the context).

> you might as well just make every version have a "p" at the end.

The regular version writes its settings to the registry. The portable version  writes to a `.ini` file. I do not want to force everyone to write settings to a `.ini` file, because some people (myself included) who don't need to/don't care about moving their settings from one computer to another (which is really the only point to run Rufus in portable mode) prefer settings to be written to the registry, especially as it means you can run different versions of Rufus from different directories, and have the settings preserved.

> The point was to not have to go to the web and download it off the page...

Again, you did not have to. You assumed that Rufus is dumb enough not to run in portable mode when it doesn't have a 'p' in its name, but you failed to realize that this is not the only parameter that tells Rufus whether it should run in portable mode or not.

People who download the portable version and let Rufus update itself have no issue with continuing to run Rufus in portable mode.

as-muncher, 2023-07-30:
ok thanks.  I didn't know that rufus updated itself - I just thought it downloaded the file.

pbatard, 2023-07-30:
Since it's a self contained single executable, Rufus updates itself by just downloading the latest `.exe`. But again, that `.exe` is smart enough to know whether it should continue to run in portable mode regardless of whether there's a `p` in its name.