pbatard/rufus #2295: Rufus not showing a device on the lists.

ItzBlazeMc, 2023-08-01:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x]  Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
There is no specific need to describe the issue in toooo much detail honestly. It's already been discussed though mine occurs with a minor difference. Rufus refuses to show me any and all devices within the device selection drop down menu. Most people go Alt + F to check the List USB drives but doesnt work for me. The usb is a small one bearing 16GB of maximum space. The USB is needed to install Windows 7 on a laptop, whilst formatting the C drive to remove the existing Ubuntu OS on it.

Im rather new to this software, still im aware enough that this might not have any linkage with any form of logs. Feel free to ask me anything  regarding this and i'll answer in the best possible way. 

With regards, and well wishes

 - Elyan

Log
---
```
<FULL LOG>
```

pbatard, 2023-08-01:
> Feel free to ask me anything regarding this and i'll answer in the best possible way.

Then please provide your log. I simply cannot help you without the log, and it's not because Rufus does not show any device that the log is useless, on the contrary.

There is no reason for you not to provide a log when you are asking for help and it is explicitly requested. It is requested because it is 100% needed __if__ you want to receive help with your issue.

pbatard, 2023-08-01:
Windows 7. Sorry but you're out of support:
https://github.com/pbatard/rufus/wiki/FAQ#user-content-Windows_XP_Windows_Vista_Windows_7_NO_LONGER_SUPPORTED

closed 2023-08-01 by pbatard

ItzBlazeMc, 2023-08-01:
The OS is Ubuntu though? If installing Windows 7 is the issue sure il change. But ive already tried it else wise

ItzBlazeMc, 2023-08-01:
aaand its ended. welp il figure it out myself.

thanks?

pbatard, 2023-08-01:
You are running Rufus on Windows 7. This is not something I can support any longer. I just don't have the scope to support OSes that have been obsoleted for years by the OS manufacturer, because trying to support currently supported OSes takes more than enough of my time.

pbatard, 2023-08-01:
It's a WINE problem. Because WINE is __not__ a Virtual Machine, it is missing the low level Windows APIs that Rufus needs to run.

WINE is __not__ Windows, and it is not even close to properly emulating Windows. It only happens to be able to run Windows application that don't do anything fancy and don't need low level Windows APIs. Rufus is not such an application.