pbatard/rufus #2296: rufus 4.2.2074 creates *.tmp files and does not delete them after closing

pult, 2023-08-02:
rufus 4.2.2074 creates *.tmp files and does not delete them after closing

pbatard, 2023-08-02:
Well, since you __chose__ not to provide a log that tells me what operation you were doing, I am unable to replicate your issue.

For one thing, if you forcefully close Rufus or it crashes, it will leave .tmp files behind, so, in the absence of any other information from you, I will assume that this is what happened.

Please remove all the `*.tmp` files you see, then run Rufus, and, if you happen to see `*.tmp` files left behind, perform the same operation you were doing and send me the log (and confirm that, after Rufus is closed, it will leave another `*.tmp`). This is the __only__ way I'm going to be able to look at the issue, as I hope you can appreciate that I'm not going to spend days creating media at random in the hope that I can replicate your issue.

pult, 2023-08-02:
not any operation, only:
1) drop all *.tmp
2)start rufus
3)close rufus

ps: rufus.ini
ShowAdvancedDriveProperties = 1
ShowAdvancedFormatOptions = 1
EnableWindowsDualUefiBiosMode = 1

pult, 2023-08-02:
not any operation, only:

1) drop all *.tmp
2) start rufus
3) close rufus

ps:

rufus.ini
ShowAdvancedDriveProperties = 1
ShowAdvancedFormatOptions = 1
EnableWindowsDualUefiBiosMode = 1

rufus\rufus.log

Rufus x64 v4.2.2074 (Portable)
Windows version: Windows 11 Pro x64 (Build 22621.1992)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0422 (uk-UA)
Will use default UI locale 0x0422
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'uk-UA'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found UAS (USB 3.1) device 'ADATA SE800 UAS Device' (174C:A89A)
Using 'autorun.inf' label for drive E: 'Ventoy-SSD'
Device eliminated because it was detected as a Hard Drive (score 21 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found

pbatard, 2023-08-02:
By the way, all the temp files created by Rufus should start with `Ruf`, so if the temp file you see is not something like `Ruf####.tmp`, chances are it is being created by another application (e.g. security solution) that happens to monitor the applications you run and creates a temp file as you run Rufus.

pult, 2023-08-02:
RufEC42.tmp

l "en-US" "English (English)" 0x0409, 0x0809, 0x0c09, 0x1009, 0x1409, 0x1809, 0x1c09, 0x2009, 0x2409, 0x2809, 0x2c09, 0x3009, 0x3409, 0x3809, 0x3c09, 0x4009, 0x4409, 0x4809
v 3.22
t MSG_001 "Other instance detected"

pult, 2023-08-02:
Same problem on other PC:
Rufus\rufus.log

Rufus x64 v4.2.2074 (Portable)
Windows version: Windows 10 Enterprise x64 (Build 19045.3208)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0419 (ru-RU)
Detected installed Windows Language Pack for 0x0422 (Ukrainian (РЈРєСЂР°С—РЅСЃСЊРєР°))
SetLGP: OpenLocalMachineGPO failed - error 80004005
Localization set to 'uk-UA'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
0 devices found

pbatard, 2023-08-02:
So it's the Rufus loc file.

And again, I am not seeing it, and this is not an issue I have ever seen on any of the PCs I tested.

For reference, we do have an explicit call to delete the loc file on exit in the code per https://github.com/pbatard/rufus/blob/master/src/rufus.c#L4138-L4139, so it's not like we somehow "forgot" to delete the file in the source.

Therefore the only circumstances where Rufus may not delete the temp loc file is:
1. If Rufus crashes on exit before it reaches the deletion call.
2. If the temp loc filename variable was somehow zeroed during execution (very unlikely, as you'd experience other issues then).
3. If the `external_loc_file` boolean variable was somehow set to `TRUE`. But if that is the case, then [you __will__ see a line `Using external loc file 'xyz.loc'` in the Rufus log](https://github.com/pbatard/rufus/blob/master/src/rufus.c#L3607-L3609), which you don't see.
4. The [`DeleteFileU()` call](https://github.com/pbatard/rufus/blob/master/src/msapi_utf8.h#L522-L532) is failing for your specific temp directory, for whatever reason.

All in all, since none of 1-3 are really viable as explanations, I'm starting to lean towards the explanation that you might have your temp directory using an uncommon path, and that this may be causing an issue with `DeleteFileU()`, because that's the only remotely logical way that can explain your issue when nobody else is seeing it.

Can you provide the full path of your temp dir?

pbatard, 2023-08-02:
Also, would you happen to use a symlink for your temp directory?
And do you experience the same issue with Rufus 4.1?

pbatard, 2023-08-02:
> User Temp folder is junction

Well, it's starting to explain things.

Please install DebugView from https://learn.microsoft.com/en-us/sysinternals/downloads/debugview and run it. Then start Rufus. You should see extra log info from Rufus startup, starting with `*** Rufus init ***` and I am curious to see if you have something like `Could not get temp directory` with an error code then.

pult, 2023-08-02:
version 4.1 does not have this problem

Rufus x64 v4.1.2045 (Portable)
Windows version: Windows 11 Pro x64 (Build 22621.1992)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0422 (uk-UA)
Will use default UI locale 0x0422
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'uk-UA'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found UAS (USB 3.1) device 'ADATA SE800 UAS Device' (174C:A89A)
Using 'autorun.inf' label for drive E: 'Ventoy-SSD'
Device eliminated because it was detected as a Hard Drive (score 21 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Found UAS (USB 3.1) device 'ADATA SE800 UAS Device' (174C:A89A)
Using 'autorun.inf' label for drive E: 'Ventoy-SSD'
Device eliminated because it was detected as a Hard Drive (score 21 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found

pbatard, 2023-08-02:
Yeah, I suspect this has to do with the code we added for people who use symlinked temp in 3e81b38c2de484beb53fa32160bb9e907d0dcc74 and complained that Rufus didn't work for them.

pult, 2023-08-02:
dbgview64.exe ( ver 4.90 ) crached wwhen capture Kernel or Passt... But worked for Capture only Win32.

pult, 2023-08-02:
dbgview rufus close:
[20064] Saved rufus.log 
[20064] SetLGP: Successfully removed NoDriveTypeAutorun policy key 
[20064] *** Rufus exit ***

pbatard, 2023-08-02:
> `[20064] Could not get actual temp directory: [0x00000001] Incorrect function.`

There we go, thanks. And that explains why people who use regular temp dirs don't see your issue.

I'll try to see if I can play with unorthodox temp dirs to replicate this issue, and potentially fix it, but this is going to be low priority for the time being.

At any rate, I appreciate your help in providing the data I need to investigate the problem.

pult, 2023-08-02:
OK. Thanks!

PS: The batch file doesn't solve the problem either:

rufus.cmd
```
set TEMP=%cd%\TEMP
set TMP=%cd%\TEMP
start rufus-p.exe
```

pbatard, 2023-08-24:
Well, I don't know how you created your Temp junction, but I am unable to replicate your issue when I used `mklink /D /H /J Temp C:\Temp` to create my junction for the `AppData\Local` Temp folder (and I can see that Rufus does properly resolve that junction to `C:\Temp` during testing).

Therefore, since this problem is unlikely to affect many people besides you (if anything, my test did validate that the changes I added to Rufus to resolve junctions for the Temp folder do work as expected when `mklink` is used correctly), and I simply don't have enough data from you on how your environment was set up, I will just close this issue.

closed 2023-08-24 by pbatard

pult, 2023-08-25:
Temp folder is junction into RAM Disk:

ImDisk Virtual Disk Driver
releases: https://sourceforge.net/projects/imdisk-toolkit/
https://github.com/LTRData/ImDisk
https://ltr-data.se/opencode.html#ImDisk