pbatard/rufus #2299: Add option on the UI for advanced users to enable Thunderbolt (non-USB) device detection

zetaloop, 2023-08-07:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I'm using a Thunderbolt drive to install windows 11 for another computer.
Rufus refused to list my drive even after turning on List USB Hard Drives.
The log said nothing about how to enable non-USB removable device support.
It was not until I enabled USB debug through the magic shortcut `Alt+.` (btw I had to try every letter before I could find this feature) that I knew I should press another magic shortcut `Alt+Ctrl+F` to list my drive. Problem solved.

How about making this option (Non-USB Hard Drives) another checkbox in advanced drive properties, with a warning popup window?
Or at least a shortcut guide or mention this (Thunderbolt drives) in FAQ?
Thanks and that will be friendlier for users who really need to do something with their Thunderbolt drives.

I know #1539 but I'm trying to make it better.

Log
---
```
Rufus x64 v4.2.2074
Windows version: Windows 11 Enterprise x64 (Build 22631.2129)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0804 (zh-CN)
Will use default UI locale 0x0804
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'zh-CN'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found non-USB removable device 'Micron_3400_MTFDKBA1T0TFH' => Eliminated
0 devices found
```
[Alt+.]
```
Processing Hub 1:
  Hub[1] = '\\?\usb#root_hub30#4&3af0ece5&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[027]: USB\VID_3277&PID_0015\5&242A2F40&0&8
  Found ID[070]: USB\VID_8087&PID_0033\5&242A2F40&0&10
Processing Hub 2:
Processing 'SCSI' device:
  Hardware ID: ''
Found non-USB removable device 'Micron_3400_MTFDKBA1T0TFH' => Eliminated
If you *REALLY* need, you can enable listing of this device with <Ctrl><Alt><F>
Processing 'SCSI' device:
  Hardware ID: ''
Found non-USB non-removable device 'NVMe SOLIDIGM SSDPFKKW020X7' => Eliminated
0 devices found
```
[Ctrl+Alt+F]
```
CAUTION: Listing of non-USB removable drives enabled
By using this unofficial cheat mode you forfeit ANY RIGHT to complain if you lose valuable data!
Processing Hub 1:
  Hub[1] = '\\?\usb#root_hub30#4&3af0ece5&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[027]: USB\VID_3277&PID_0015\5&242A2F40&0&8
  Found ID[070]: USB\VID_8087&PID_0033\5&242A2F40&0&10
Processing Hub 2:
Processing 'SCSI' device:
  Hardware ID: ''
Found non-USB removable device 'Micron_3400_MTFDKBA1T0TFH'
Processing 'SCSI' device:
  Hardware ID: ''
Found non-USB non-removable device 'NVMe SOLIDIGM SSDPFKKW020X7' => Eliminated
1 device found
Windows volumes from this device:
● \\?\Volume{6ac9afcb-6ba9-4155-a589-953ebd79b38d}\ @1048576
Disk type: FIXED, Disk size: 1TB, Sector size: 512 bytes
Cylinders: 124519, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {EAE66245-4500-453C-9716-6CDBA04F7813}
Max parts: 128, Start Offset: 17408, Usable = 1024209508864 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Basic data partition'
  Detected File System: NTFS
  ID: {6AC9AFCB-6BA9-4155-A589-953EBD79B38D}
  Size: 953.9 GB (1024208477696 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
```

pbatard, 2023-08-07:
I hear you, and that's something I would have added _if this could be accomplished without devoting a lot of resources into it (and still ending up with an imperfect result)_, but the problem is that detecting Thunderbolt connections is a non-straightforward task, because, per design, Thunderbolt masquerades as regular PCI Express, which of course makes a lot of sense, but then makes it next to impossible to differentiate between an __internal__ disk device, that is connected directly through the internal PCI Express bus of your motherboard, which I really don't want people to be able to access and format with Rufus, and and external disk device connected through Thunderbolt.

In short, unlike what is the case for USB or VHD drives, when querying disk devices, there is no direct property you can obtain from Windows that tells you _"Hey, this is a Thunderbolt device"_. All you get instead is: _"Hey, this is yet another disk device that is connected through PCI Express"_.

So, for Rufus to be able to list Thunderbolt devices, I would have to maintain a list of all the current Thunderbolt controllers hardware IDs, then perform a traversal to the root PCI Express parent to find out if there's any hardware ID that matches the one from my list (because unfortunately, you need to go more than a few levels up to get those IDs on account that, at least on the systems I tested, the direct parent or grandparent are just generic Windows PCI Express switch ports [which are not a feature of Thunderbolt and can exist internally on any modern system], so you need to go at least 3-4 levels up before you can get to a parent element that may match a known Thunderbolt controller ID).

But the main issue is that, even after you add a search for hardware IDs up to the PCI Express Root (which is doable, but would add its fair amount of complexity and caveats to the code), you also need to keep an up to date (and working) list of Thunderbolt controller IDs, which is it's own PITA to maintain. Sure, I could try to add a script that parses the `pci.ids` list from https://pci-ids.ucw.cz/ and adds everything that has Thunderbolt in its name, but this also adds its fair share of extra work and as a developer, you have to ask yourself the question: _"Will the benefits of doing all this extra work actually outweigh the costs?"_

Unfortunately for you, at this stage, my answer to this remains (since this is something I have been considering for a while) __no, the cost is simply not worth the benefits__.

And the main reason that made me tip towards the _"it's not worth the development cost at this stage"_ is that __most__ Thunderbolt disk enclosures make sure to set a __removable__ attribute, so that <kbd>Ctrl</kbd>-<kbd>Alt</kbd>-<kbd>F</kbd> should just work.

Sadly, it appears that you ran into a corner case where the enclosure you got does not set the removable flag:
```
Found non-USB non-removable device 'NVMe SOLIDIGM SSDPFKKW020X7' => Eliminated
```

So, while I agree that, in the absolute, your issue would be fixed if Rufus could just detect Thunderbolt connected disks and list them, part of your problem is that you are using a Thunderbolt enclosure (or possibly a Thunderbolt controller) that was not engineered properly because, __obviously__, anything attached through a Thunderbolt connection should be declared as removable, and, with your specific hardware environment, this is not the case.

For instance, this is what I get on an Intel NUC system with an Intel Tiger Lake Thunderbolt controller and an Orico Thunderbolt to SSD enclosure:
```
Found non-USB removable device 'TS128GMTE110S' => Eliminated
If you *REALLY* need, you can enable listing of this device with <Ctrl><Alt><F>
```

And because the device is properly detected as __removable__, I can then use <kbd>Ctrl</kbd>-<kbd>Alt</kbd>-<kbd>F</kbd> so that Rufus can then use that device:
```
CAUTION: Listing of non-USB removable drives enabled
By using this unofficial cheat mode you forfeit ANY RIGHT to complain if you lose valuable data!
Found non-USB removable device 'TS128GMTE110S'
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x81
Disk type: FIXED, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15566, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 0
Disk GUID: {3D9F0910-0A8D-4248-A5F3-B7404DE10101}
Max parts: 128, Start Offset: 17408, Usable = 128035641856 bytes
```

So this is the main reason why I currently do not see going out of my way to detect Thunderbolt devices as a good investment of my time, because most Thunderbolt disk devices _should_ become accessible with <kbd>Ctrl</kbd>-<kbd>Alt</kbd>-<kbd>F</kbd>, and if it isn't, then I would say that part of the blame lies with whoever manufactured your hardware and forgot to make sure that the Thunderbolt disks listed by Windows will be seen as removable.

For reference, Rufus detects the "removeability" of a disk by [querying the __standard__ Windows hardware property `SPDRP_REMOVAL_POLICY`](https://github.com/pbatard/rufus/blob/5084317dd7fc58fb62d302b8f9239f6834e60714/src/dev.c#L692-L693), so there's nothing really fancy going on there: it's a direct query of a property that should be set by the Thunderbolt hardware.

Now, with USB 4.0 and its integration of Thunderbolt, we may review our current cost/benefit analysis, and go through a whole trying to detect if a device is connected through a USB 4.0/Thunderbolt controller, but I'm afraid that, for now, even as I understand that it is inconvenient for the specific hardware you are using, I am not planning to do more for Thunderbolt connected devices than relying on <kbd>Ctrl</kbd>-<kbd>Alt</kbd>-<kbd>F</kbd>.

As a result, I will close this request.

closed 2023-08-07 by pbatard

zetaloop, 2023-08-08:
> Sadly, it appears that you ran into a corner case where the enclosure you got does not set the removable flag:
> 
> ```
> Found non-USB non-removable device 'NVMe SOLIDIGM SSDPFKKW020X7' => Eliminated
> ```
Sorry, but this Solidigm disk is not the Thunderbolt device. It is my boot drive.
`Micron_3400_MTFDKBA1T0TFH` <-- This is the Thunderbolt drive.
Actually, the List Non-USB Hard Drives option worked quite well for me.

Thank you for all your great work. If it is hard to add the UI option, could you please add a comment about how to enable support for Thunderbolt drives in FAQ? Then it will be easier for users to find a solution (although unwarranted) at least.