pbatard/rufus #2300: Where can i see what rufus change to the windows 11 iso file

glda19, 2023-08-07:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

glda19, 2023-08-07:
Where can i see what Rufus dos with the windows 11 file To let windows 11 install on non supported pc

pbatard, 2023-08-07:
Do a search for `HKEY_LOCAL_MACHINE\SYSTEM\Setup\LabConfig` and `BypassTPMCheck`, `BypassSecureBootCheck`, `BypassRAMCheck`. That'll point you to articles that describe what Rufus does to install Windows 11 on unsupported PC.

See also https://github.com/pbatard/rufus/blob/5084317dd7fc58fb62d302b8f9239f6834e60714/src/wue.c#L750-L927.

closed 2023-08-07 by pbatard