pbatard/rufus #2302: Add .wic extension to disk images filter

warfery, 2023-08-10:
I am using .wic images and rufus can burn it without any problem. Only thing the .wic is not between list of supported images and I need to change filter to *.*. Is it possible to add .wic extension to images extension list? Thanks.

pbatard, 2023-08-10:
If the `.wic` extension was reported by Google as something commonly associated with image files, I might.

But right now, [Google says](https://www.google.com/search?q=.wic+extension) that `.wic` is commonly associated with a wavelet compression algorithm, which tells me that it's not a common image extension, and as a result, I am not planning to add it to the list.

You also chose not indicate what application creates those `.wic` files or where you obtained them, which also doesn't help your case.

As such you will continue to have to use `*.*` if you want to work with `.wic` file and I will close this issue.

closed 2023-08-10 by pbatard

warfery, 2023-08-11:
Sorry, .wic is disk image format produced by wic tool from Yocto project.
More info here:
https://docs.yoctoproject.org/dev-manual/wic.html#
https://stackoverflow.com/questions/49527057/difference-between-wic-and-hddimg-format-in-yocto

pbatard, 2023-09-08:
Note that, since someone else also made a request to add `.wic` to the defaults, I have now added it in [99bffe836424bec147972b3691898a4036083284].

I'm still a bit dumbfounded as to why the Yocto folks had to introduce their own extension where the __de-facto__ `.img` would have worked the same, but hey...