pbatard/rufus #2306: Feature request: Copy files from sources\$OEM$ to their folders

vichingo455-temp, 2023-08-16:
﻿Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
This can be useful as also the Windows Setup does that.

Examples:
$$ is %windir%
$1 is %systemdrive%

Log
---
```
<FULL LOG>
```

pbatard, 2023-08-16:
I don't understand your question.

The Windows Setup is what runs on the media created by Rufus, so it is going to copy any file you have in `sources\$OEM$` to the drive.

All you need to do then is create that `sources\$OEM$` yourself and populate it with content.

Or are you talking about Windows To Go? Because that is not a feature I am planning to add to Windows To Go, since most people would work with official Windows ISOs and, there again, you can just add the OEM content yourself.

vichingo455-temp, 2023-08-16:
oh sorry, I forgot a detail. I meant for the Windows To Go thing in Rufus

pbatard, 2023-08-16:
I design Rufus for the general use case (people using officially downloaded ISOs from Microsoft) not for corner cases (people using Windows ISOs sourced from third parties __AND__ unaware of what comes with those ISOs __AND__ creating a Windows To Go drive from these ISOs).

See https://github.com/pbatard/rufus/wiki/FAQ#user-content-Why_cant_you_just_add_the_feature_I_want.
Your case falls into 2.