pbatard/rufus #2308: remove Rufus update requirement to download

1268, 2023-08-19:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.

Issue description
-----------------

I see this:

<https://github.com/pbatard/rufus/wiki/FAQ#Help_I_dont_see_the_ISO_download_button>

but I do not agree with the premise. in my situation, I DO NOT want Rufus to ever check for updates for itself, but I DO want to download Windows with Rufus. So can we remove this requirement and just always allow download, regardless of update setting?

pbatard, 2023-08-19:
> but I do not agree with the premise.

Then I hope you can understand that there is nothing I can do for you. I am not planning to dissociate the check for update from the ISO download because, from where I stand, people who want one but not the other are making either an irrational or an uneducated decision. I chose to associate the two because the behaviour of going over the network to run something that is downloaded from a remote server entails the same risks and rewards.

Plus, if you chose to disable the check for update, I personally consider that you are doing yourself a major disservice (because I am not upgrading the application for the sake of updates, but because there are improvements to apply and issues to address) as well as distrusting my ability to provide users with updates in a safe fashion (for more on this see https://github.com/pbatard/rufus/wiki/Security). So, while I do want users to ultimately have the choice of whether they want to enable the application to connect to the internet (which is __really__ what the check for updates choice is all about), you need to understand that this what the check for updates choice is all about and that, as a result, the ISO download feature has to be tied to that choice, and be disabled if you do disagree with the application connecting to the internet to check for new versions.

closed 2023-08-19 by pbatard

1268, 2023-08-19:
> people who want one but not the other are making either an irrational or an uneducated decision.

Its a rational decision. having tool that does updates daily/weekly/monthly, is much different from a one time download of an operating system.

> I chose to associate the two because the behaviour of going over the network to run something that is downloaded from a remote server entails the same risks and rewards.

see above. a one time download has less risk that a repeated download of arbitrary data.

> Plus, if you chose to disable the check for update, I personally consider that you are doing yourself a major disservice

I only reinstall my operating system about once every four years, so all this current restriction did is make my experience confusing and frustrating. there is nothing technical that makes the updates mandatory, its more a personal decision by the maintainer.

> as well as distrusting my ability to provide users with updates in a safe fashion

yes, I do distrust in that regard. its nothing personal, I dont really trust any updates. this is why I prefer to update manually, so that I can review any updates and make my own decision to install or not.

> So, while I do want users to ultimately have the choice of whether they want to enable the application to connect to the internet

thats the thing. this is not a question of "connect to the internet". the "connect to the internet" question is only valid in the context OF UPDATES. people have valid concerns about programs randomly doing telemetry or whatever else. this is COMPLETELY DIFFERENT from a user EXPLICITLY requesting a single download of an ISO file.

> the ISO download feature has to be tied to that choice, and be disabled if you do disagree with the application connecting to the internet to check for new versions.

see previous comment. essentially a link has been created where there is none. some might personally might feel its needed to have both or none, but I would argue its a valid use case to disconnect them.

pineapple63, 2023-08-19:
The ISO download function does kinda download an “update”, as it is designed such that the download feature can be updated without having to update the main executable (to allow new downloads to be added, or to allow the feature to be fixed if Microsoft decides to break it again)

And the check for updates feature only runs when you launch Rufus, and even then, it will ASK if you want to download the new version

pbatard, 2023-08-19:
>  having tool that does updates daily/weekly/monthly, is much different from a one time download of an operating system.

You'll have to explain how having a tool that issue a less than 1 KB transfer every month if you choose so is that annoying to you.

> see above. a one time download has less risk that a repeated download of arbitrary data.

I'm gonna stop right there because you validated exactly the point I was making above (and also are indicating that you did not read the link I gave you) about __irrationally__ distrusting my ability as an application developer to provide you with a safe means of updating the application.

Furthermore, you seem to completely miss the fact that the ISO download feature __does execute a rather lengthy third party PowerShell script in elevated mode__, whereas all the check for update does is download [this file](https://github.com/pbatard/rufus-web/blob/gh-pages/Rufus_win.ver).

Thus, you have both demonstrated that you misunderstand what the ISO download feature really does (please have a look [here](https://github.com/pbatard/Fido), and why it is far from being as innocuous as you think, and that you taking an irrational decision as a result.

> this is not a question of "connect to the internet". 

It is.

> people have valid concerns about programs randomly doing telemetry or whatever else.

Then read the bloody Security Page, as you may [learn a thing or two](https://github.com/pbatard/rufus/wiki/Security#user-content-Data_Collection_and_User_Consent) that will avoid making an uneducated decision.

1268, 2023-08-19:
another good argument is this:

currently one can have auto updates without every downloading an ISO, but one cannot download an ISO without auto updates. and again, requiring Rufus updates is not a technical requirement to downloading an operating system, merely a personal decision by the maintainer.

but because I disagree with this personal decision, I am marked as irrational. oh well. if someone wants to attack me rather that the merit of the argument, [they are welcome to do so](https://wikipedia.org/wiki/Ad_hominem)

pbatard, 2023-08-19:
An _ad hominem_ is an attack on the assumed character of a person without providing evidence.

I have provided evidence that you did not read the Security page and __therefore__ that your argument is irrational, because it is lacking the amount of data required to be educated and receivable.

1268, 2023-08-19:
> I have provided evidence that you did not read the Security page and **therefore** that your argument is irrational, because it is lacking the amount of data required to be educated and receivable.

is anyone willing to address the point I have made multiple times about the technical requirement? the avoidance thus far does not look good, and only seems to reinforce the point that linking the two types of downloads is merely a personal preference of the maintainer, rather than a technical requirement.

pbatard, 2023-08-19:
There's only so many way I can reiterate what I already explained to you as to why it does make __logical__ sense to have the two linked once you understand what each feature entails and what the initial choice about the check for update is about.

As a result, I will lock the conversation. Feel free to fork the source and create your own version of Rufus that dissociates the check for update from the download feature.