pbatard/rufus #2310: I Think Rufus Killed My USB Pendrive!

cfsakil, 2023-08-22:
Yesterday i wanted to make my pendrive bootable so i could use Ubuntu os on it. So i followed a popular youtube tutorial and followed step by step.

I've downloaded 
OS: ubuntu-22.04.3-desktop-amd64
Rufus: rufus-4.2.exe

Then i opened Rufus and select my Pen drive and os., And used 70% on data persistence. 
Then i click on Flash. And it was progressing. Maybe 30% + then appear a failure message, That was like Failed to flash or something else. So i tried click on Flash Button again but that was disabled. Then i searched on google for fixing the issue. So i found out aboud cmd "Diskpart-> List disk" but i saw my Pendrive shows Total size 0, Used 0. So i open File Explorer. Then Right click and hit "Format". 
Then after laoding its failed. 

After that in cmd, Disk managment no where i find my pendrive anymore not even in Rufus.
When i put usb i hear the sound. And if i go explorar its appear "USB Drive (H)" That time rufus not responding. And if i click to open that drive by File its loading and thats Not responding too. After few second its start working but USB Drive (H) is not there anymore. And i got notification like this "Windows cant recognise". My USB Pendrive is not found no where except Device manager. There shows Unknown USB Device (Device Descriptor request failed).  

There is any way to fix this please? :)))

winapiadmin, 2023-08-24:
Let's analyze the problem.

1. "Pendrive shows Total size 0, Used 0."

It always mean there isn't have a valid filesystem.

2. "Then Right click and hit "Format".
Then after laoding its failed."

It may be the physical errors (e.g, bad sectors) in the drive.

3. "And i got notification like this "Windows cant recognise". My USB Pendrive is not found no where except Device manager. There shows Unknown USB Device (Device Descriptor request failed)."

The USB may be errors, or the USB port. Did you put another USB devices and successfully?

4. "NOTE: This device is a USB 3.0 device operating at lower speed..."

It may be the problem with USB drive detection, or maybe itself(that pendrive).

5. "Could not get device number for device \.\H: [0x000001B1] A device which does not exist was specified.
Could not get Disk Extents: [0x000001B1] A device which does not exist was specified.
Ignoring volume '\?\Volume{2cda9e9d-40c6-11ee-bd2f-18c04df83214}' because it has more than one extent (RAID?)...
Could not open \?\Volume{6341ebf9-3f4f-11ee-bd28-18c04df83214}: [0x00000002] The system cannot find the file specified.
Ignoring drive '\.\G:' as it spans multiple disks (RAID?)
Could not open \.\PhysicalDrive2: [0x00000002] The system cannot find the file specified.
Could not open \.\PhysicalDrive2: [0x00000002] The system cannot find the file specified.
Ignoring volume '\?\Volume{2cda9e9d-40c6-11ee-bd2f-18c04df83214}' because it has more than one extent (RAID?)...
No volume information for drive 0x82
Could not open \.\PhysicalDrive2: [0x00000002] The system cannot find the file specified.
No file system is selectable for this drive
0 devices found"

According to 1. , 2. , 3. , that's why the drive H is not exist. It's also why no filesystem is selectable for this drive.
And it's also solve the "RAID drive" problem.

pbatard, 2023-08-24:
> And it was progressing. Maybe 30% + then appear a failure message,

That's because:

1. Your drive failed with an internal hardware failure that Rufus had nothing to do with (the flash memory used in consumer USB flash drives is a lot more unreliable than people imagine and will fail sooner rather than later).
2. __As a result of this internal failure__ Rufus was unable to complete its operations and reported an error.

> There is any way to fix this please? :)))

From your description of the issue, that is unlikely.

You can try a different OS (e.g. Linux) to see if it can access and repartition/reformat your drive, but most likely you will also find that your drive is not accessible there either.

Since there is really nothing Rufus can do about hardware errors, especially ones that are reported by Windows itself, I will close this issue.

closed 2023-08-24 by pbatard

Hrxn, 2023-08-24:
> 
> 1. Your drive failed with an internal hardware failure that Rufus had nothing to do with (the flash memory used in consumer USB flash drives is a lot more unreliable than people imagine and will fail sooner rather than later).

Please don't crucify me for asking this question here, but I just realized that maybe you are one of the folks that actually *can* be asked this question, because they are actually knowledgeable in this specific subject.

Are there specific flash drives that you would recommend? Types, manufactures, whatever?
Or maybe drives you explicitly do *not* recommend?
Because I'm using a normal "consumer" USB pen drive, and I backup my important stuff there, and I've been worrying about this exact issue for quite some time..

pineapple63, 2023-08-24:
> Please don't crucify me for asking this question here, but I just realized that maybe you are one of the folks that actually _can_ be asked this question, because they are actually knowledgeable in [...]

If you were following best practice for backups, then i would say that it shouldn’t be an issue (the advice ive heard is to have at least THREE copies of the data, on TWO different storage mediums, such as Hard Drive/Magnetic media, SSD/Flash Storage, Optical media, Cloud storage)

From my experience, it seems that cheaper USB sticks are more likely to fail
Sandisk Dual Drive USBs seem to be ok (ive got a few of them, and haven’t (yet) had one fail)