pbatard/rufus #2312: Rufus doesn't create the perfect USB for Kali Live, starting 2023.3

UberGeekDude, 2023-08-24:
I just downloaded the latest kali Live ISO and used Rufus 4.2 to create the Live Boot USB.
Before this release, Rufus worked flawlessly but Kali.org changed something up.  (Not sure what)

**Now the live Boot USB only boots to grub.**

I used the default settings MBR Partition, BIOS or UEFI FAT32 and 8k byte cluster size.
[image]

Kali.org Documents now recommends another tool, Etcher, which uses dd to create the partitions.  I know I have that dd option in Rufus but the default has worked great for many years, and with persistence partition already setup.  And I can edit the files (grub) with FAT32 partition.

Current ISO: https://www.kali.org/get-kali/#kali-live
https://cdimage.kali.org/kali-2023.3/kali-linux-2023.3-live-amd64.iso
SHA256sum: 7755f44318e6183d8411128b8a56c2982dbf0dd4c62870f4e5ca28c6df73fd9a

pbatard, 2023-08-25:
Thanks for the report.

I am able to replicate the issue with UEFI boot but, unfortunately, there is nothing I can do in Rufus to fix this, since this is a problem with Kali having broken File System Transposition support on their source ISOs (despite being based on Debian, which does properly support File System Transposition).

For reference, [File System Transposition support](https://www.mail-archive.com/grub-devel@gnu.org/msg34346.html) is:

> the ability to take the content of a UEFI bootable media and copy it, at the file system level, to a partition that was 
independently created and formatted by the user, while preserving the ability of the media to boot in UEFI mode.

It's actually a bit surprising that, for a security oriented distro, Kali happened to break FST, because FST is what provides users with the ability to create bootable media __without using ANY third party tools, regardless of the OS you use__, which means that, with proper FST compatible media, you shouldn't even need something like Rufus (or balenaEtcher) on Windows to create a Kali USB drive.

At this stage, I have no idea what Kali actually did to break FST, but I suspect that they must have removed something like a GRUB file system module like FAT (but I'm not entirely sure about this, because FAT is pretty much always needed to access the ESP), with the end result that GRUB is unable to access its configuration files and bails out.

I had a quick look at the [recent commits from Kali's GRUB source](https://gitlab.com/kalilinux/packages/grub2/-/commits/kali/master) but I don't see anything obvious, especially as they seem to be trying to be following Debian's GRUB, and Debian releases (I checked again with the [latest Debian Testing](https://wiki.debian.org/DebianTesting)) don't have the issue.

If you are a Kali user, I would report this regression to the Kali distro maintainers, because it is theirs to fix, as the Kali folks will probably understand that dropping FST support to force their uses to only ever use DD mode for media creation, is actually not beneficial, on account that, besides requiring the use of third-party tools on Windows, DD does have limitations and caveats (for instance, using DD, you cannot do things like single media, whereas having support for FST is for instance what [allows Debian ARM64 to be installed using a single media on the Raspberry Pi](https://forums.raspberrypi.com//viewtopic.php?f=50&t=282839), which can be very beneficial on Single Board Computers with limited I/O slots) is declare Kali, or, more crucially, allows users to add the necessary non-free WiFi firmware blobs that they might require for network connectivity without the need for an extra drive).

Otherwise, if Kali doesn't fix this, I will have to force Kali ISOs to always use DD in the next version of Rufus, which I am already doing for a couple other distros whose maintainers are either under the false impression that DD is a panacea, or are annoyed at the mimimal amount of extra work they need to accomplish to ensure FST compatibility, but which of course, means that you will no longer be able to create persistent Kali partitions from Rufus.

UberGeekDude, 2023-08-25:
Thanks.  I have already grunted my frustrations in their Discord forum.  People's response was "read the docs".
I will try Rufus again, with dd.  The biggest problem I have is changing grub.cfg/config.cfg to set a default boot (persistence) and timeout, plus get rid of that awful boot beep.

set default=3
set timeout=2
# comment out play tones.

Currently, I am trying to learn how to edit grub via the command prompt.  I am a bit lost with that, so I need more research.
I realized I can plug the USB into another Linux system, mount the drive and edit those files.
It was so easy to edit the FAT partition in Windows, before removing the USB for my USB Live boot sessions.

Thank you for your work on Rufus, to make it an awesome go-to tool for what I need.
Regards.

closed 2023-08-25 by UberGeekDude

pbatard, 2023-08-25:
>  The biggest problem I have is changing grub.cfg/config.cfg

Won't be able to do that with DD. From what I can tell, DD uses a read-only ISO-9660 partition for the config files, so you won't be able to edit the config files until a [bug is logged with Kali](https://www.kali.org/docs/community/submitting-issues-kali-bug-tracker/) to let them know that they broke FST.

And don't bother with Discord. Distro maintainers read bug reports not Discord, especially when, on user-led Discord, the default behaviour seems to point the finger at anything that resides outside your established community as the reason for all your woes, even when it can be easily demonstrated (again, just compare with what Debian does) that there exists something that Kali appears to have broken and should/could fix.

pbatard, 2023-08-30:
For the record, I submitted [BUG 8441](https://bugs.kali.org/view.php?id=8441) to the Kali people with regards to this issue.