pbatard/rufus #2314: Extraction Error while creating a bootable USB

sharjh, 2023-08-28:
﻿
Checklist
---------
- [✓] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [✓] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [✓] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [✓] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
while making a bootable drive for EndeavourOS installation i always run into an extraction failure which happens midway through the process right after rufus says 'READY' before it even completes, I previously used this drive to install linux mint which worked completely fine.
I tried selecting GPT and DD mode but nothing works and results in the same error
I hear the usb ejection sound before it happens which might be the cause of it
The log attached to this issue uses the method described in EndeavourOS website

Log
---
```
Rufus x64 v4.2.2074
Windows version: Windows 11 Home Single Language x64 (Build 22621.2134)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
Using 'autorun.inf' label for drive E: 'EOS_202308'
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3891, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 29.8 GB (32009879552 bytes)
  Start Sector: 2048, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/syslinux/syslinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/pre2-11-gbf6db5b4 (from '/syslinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'EOS_202308'
  Size: 1.8 GB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: Endeavouros_Cassini_Nova-03-2023_R3.iso (1.9 GB)

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Opened \\.\PhysicalDrive1 for exclusive write access
Requesting logical volume handle...
No logical drive found (unpartitioned?)
+++++++++++++++++++++++++
Write error at sector 1572864: [0x000001B1] A device which does not exist was specified.
Retrying in 5 seconds...

Write error at sector 1572864: [0x00000037] The specified network resource or device is no longer available.
Retrying in 5 seconds...

Write error at sector 1572864: [0x00000037] The specified network resource or device is no longer available.
Retrying in 5 seconds...

Write error at sector 1572864: [0x00000037] The specified network resource or device is no longer available.
Could not open \\.\PhysicalDrive1: [0x00000002] The system cannot find the file specified.
Could not open \\.\PhysicalDrive1: [0x00000002] The system cannot find the file specified.

0 devices found
rufus_error_log.log
3 KB
```

sharjh, 2023-08-29:
another attempt

```
Rufus x64 v4.2.2074
Windows version: Windows 11 Home Single Language x64 (Build 22621.2134)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
Using 'autorun.inf' label for drive E: '32 GB'
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3891, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 29.8 GB (32009879552 bytes)
  Start Sector: 2048, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/syslinux/syslinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/pre2-11-gbf6db5b4 (from '/syslinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'EOS_202308'
  Size: 1.8 GB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: Endeavouros_Cassini_Nova-03-2023_R3.iso (1.9 GB)

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 29.8 GB)
Waiting for logical drive to reappear...
Formatting to FAT32 (using IFS)
Using cluster size: 16384 bytes
Quick format was selected
Creating file system...
Format completed.
Opened \\.\PhysicalDrive1 for exclusive write access
Writing Master Boot Record...
Using Rufus protective MBR
Writing protective message SBR
Found volume \\?\Volume{477438fa-463c-11ee-af2d-50c2e88eeaaf}\
Successfully remounted \\?\Volume{477438fa-463c-11ee-af2d-50c2e88eeaaf}\ as E:
Extracting files...
Image is an ISO9660 image
This image will be extracted using Joliet extensions (if present)
Extracting: E:\EFI\BOOT\BOOTx64.EFI (90 KB)
Extracting: E:\arch\boot\amd-ucode.img (60 KB)
Extracting: E:\arch\boot\intel-ucode.img (7.0 MB)
Extracting: E:\arch\boot\licenses\amd-ucode\LICENSE.amd-ucode (3.7 KB)
Extracting: E:\arch\boot\licenses\intel-ucode\LICENSE (1.6 KB)
Extracting: E:\arch\boot\licenses\memtest86+\license.txt (17.7 KB)
Extracting: E:\arch\boot\memtest (141.0 KB)
Extracting: E:\arch\boot\x86_64\initramfs-linux.img (91.5 MB)
Extracting: E:\arch\boot\x86_64\vmlinuz-linux (12.1 MB)
Extracting: E:\arch\grubenv (1 KB)
Extracting: E:\arch\pkglist.x86_64.txt (17.1 KB)
Extracting: E:\arch\version (11 bytes)
Extracting: E:\arch\x86_64\airootfs.sfs (1.7 GB)
Write error [0x000003EE] The volume for a file has been externally altered so that the opened file is no longer valid.
Retrying in 5 seconds...
Write error [0x000003EE] The volume for a file has been externally altered so that the opened file is no longer valid.
Retrying in 5 seconds...
Write error [0x000003EE] The volume for a file has been externally altered so that the opened file is no longer valid.
Retrying in 5 seconds...
Write error [0x000003EE] The volume for a file has been externally altered so that the opened file is no longer valid.
  Error writing file: [0x000003EE] The volume for a file has been externally altered so that the opened file is no longer valid.
Extracting: E:\loader\entries\archiso-x86_64-linux-nv.conf (423 bytes)
  Unable to create file: [0x00000003] The system cannot find the path specified.
Unable to create E:\syslinux.cfg - booting from USB will not work

0 devices found
```

pbatard, 2023-08-29:
I'm afraid I am unable to replicate your issue:

<details>
<summary>Click to show log</summary>

```
Rufus x64 v4.2.2074
Windows version: Windows 11 Pro for Workstations x64 (Build 22621.2215)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB device 'SanDisk Extreme USB Device' (0781:5580) [ID]
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x89
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1946, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x1BDB3CC0
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI System Partition (0xef)
  Detected File System: FAT16
  Size: 113 MB (118489088 bytes)
  Start Sector: 3753600, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/syslinux/syslinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/pre2-11-gbf6db5b4 (from '/syslinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'EOS_202308'
  Size: 1.8 GB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: Endeavouros_Cassini_Nova-03-2023_R3.iso (1.9 GB)

Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'F:' as volume mountpoint
Opened \\.\PhysicalDrive9 for exclusive write access
Requesting logical volume handle...
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Opened \\.\PhysicalDrive9 for shared write access

Found USB device 'SanDisk Extreme USB Device' (0781:5580) [ID]
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x89
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1946, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x1BDB3CC0
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI System Partition (0xef)
  Detected File System: FAT16
  Size: 113 MB (118489088 bytes)
  Start Sector: 3753600, Boot: No
Checking for Rufus updates...
Checking release channel...
No new release version found.
Checking beta channel...
No new beta version found.
```
</details>

Also, the thing that seems most relevant to troubleshooting (trying with a different drive and trying on a different computer) are the exact two things you haven't tried:

> - [x] I also tried one or more of the following:
>   - [ ] Using a different USB drive.
>   - [x] Plugging the USB into a different port.
>   - [ ] Running Rufus on a different computer.

All in all, it looks like either an issue with your specific computer or with the specific flash drive you are using and, since you are the only person to report these kind of issues, I'm afraid I have to dismiss it as an environmental/hardware issue.

The one loosely related issue I know is that, when a drive contains an ESP (EFI System Partition), Windows can sometimes be reluctant about granting access, but these are usually transient errors, that can be alleviated by zeroing the drive (using <kbd>Alt</kbd>-<kbd>Z</kbd> in Rufus) for about a minute and then cancelling.

But seeing how you consistently appear to experience errors, even on drives that have no ESP, I'm afraid there is not much I can do. It really does seem related to the specific environment/hardware you use.

sharjh, 2023-09-05:
it is in fact, a drive issue