pbatard/rufus #2315: ISO download window does not open when mapped drives from the search path are not available

Tom-EllisEVENTS, 2023-08-31:
This is a followup to closed issue #2030 

Apparently, the fix [from this commit](https://github.com/pbatard/Fido/commit/ed1a6b31a61574ae95f501b80a254ad1ff7798e3)  does not work anymore, because the "-IgnoreWarnings" parameter that has been added got lost again sometimes later.

Using **_WarningAction = "Ignore"_** seems **not to be equivalent** to using **-IgnoreWarnings**

After adding the following line right after "WarningAction", it works again:

IgnoreWarnings       = $true

Full context:

```
$Signature = @{
	Namespace            = "WinAPI"
	Name                 = "Utils"
	Language             = "CSharp"
	UsingNamespace       = "System.Runtime", "System.IO", "System.Text", "System.Drawing", "System.Globalization"
	ReferencedAssemblies = $Drawing_Assembly
	ErrorAction          = "Stop"
	WarningAction        = "Ignore"
	IgnoreWarnings       = $true
	MemberDefinition     = @"
		[DllImport("shell32.dll", CharSet = CharSet.Auto, SetLastError = true, BestFitMapping = false, ThrowOnUnmappableChar = true)]
		internal static extern int ExtractIconEx(string sFile, int iIndex, out IntPtr piLargeVersion, out IntPtr piSmallVersion, int amountIcons);

		[DllImport("user32.dll")]
		public static extern bool ShowWindow(IntPtr handle, int state);
		// Extract an icon from a DLL
		public static Icon ExtractIcon(string file, int number, bool largeIcon) {
			IntPtr large, small;
			ExtractIconEx(file, number, out large, out small, 1);
			try {
				return Icon.FromHandle(largeIcon ? large : small);
			} catch {
				return null;
			}
		}
"@
}

```

pbatard, 2023-09-08:
Thanks for the report and apologies for the late reply!

What you are describing, along with your analysis, makes sense, so I will apply your patch shortly.

closed 2023-09-08 by pbatard

Tom-EllisEVENTS, 2023-09-12:
Great! Thanks a lot!