pbatard/rufus #2316: Get the following error message on screen

Apache060, 2023-09-01:
`Hello,`
I have the following issue rufus  I can burn the iso file to usb stick, but when I try to boot from the USB stick the following message appears on the screen.
Please see screen shot belo 

[image]

JonnyTech, 2023-09-01:
The detailed message on screen has clear instructions - I suggest that you read and follow them.

pbatard, 2023-09-01:
See also this [very relevant FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#user-content-ERROR_BIOSLEGACY_BOOT_OF_UEFIONLY_MEDIA).

Since you chose not to provide a log of how you created the media, despite being very explicitly asked to do so, there really is no extra advice we can provide.

closed 2023-09-01 by pbatard

Apache060, 2023-09-07:
Hello,
The error message is gone, the issue now is that the pc will say booting from Device then the screen goes blank and all I get is a white dash flashing on the screen.

Apache060, 2023-09-08:
The issue is now fixed.

Thanks for your help