pbatard/rufus #2317: ssd not working when using rufus and rebooting says raw partition

kamikazilucas, 2023-09-02:
when i install rufus as windows to go then reboot pc it says the drive is not accesible and that is is corrupted or unreadable, when i go into partitions the parition is a raw, how do i fix

pbatard, 2023-09-02:
That looks like a pure Windows or hardware issue, not a Rufus issue.

At any rate, since you did not provide any of the data that was asked of you, there's not much advice we can provide and as a result I will close your issue.

closed 2023-09-02 by pbatard

kamikazilucas, 2023-09-02:
well how come it works fine but as soon as i install rufus it gets corrupted

pbatard, 2023-09-02:
Without the log, which you __CHOSE NOT TO PROVIDE DESPITE BEING EXPLICITLY ASKED TO DO SO__ I have no way to even start trying to answer your question. Also, Rufus does not "install". You create a Windows bootable media with Rufus using a Windows ISO. You do not "install" Rufus.

kamikazilucas, 2023-09-02:
[rufus.log](https://github.com/pbatard/rufus/files/12504212/rufus.log)
here is log, im aware it does not install but it does activate and change the drive into something else

pbatard, 2023-09-02:
Still doesn't tell me how you created your media.

All I can see from the log you provided is that you appear to have a properly partitioned Windows To Go drive, and I still don't understand what your issue is.

On one hand, you say _"when using rufus and rebooting says raw partition"_ which would imply that, after you reboot Windows (the one where you ran Rufus) and plug your Windows To Go SSD again, Windows Disk Manager says that the drive is unpartitioned. But this is contradicted by the log you provided, where Rufus does see the partitions, so Disk Manager should see them too.

But then you say Rufus _"does activate and change the drive into something else"_, which does not mention rebooting (and for the record, "activation" is a word reserved for Windows license registration. There's no such thing as Rufus "activating" anything, as Rufus does not deal with licensing).

So can you please describe your issue more precisely?
- Is the drive visible in disk manager and accessible in Windows before you use Rufus?
- What exactly do you do in Rufus? I doubt you are just running Rufus and closing it for no reason. So can we please have the log of what it is you do in Rufus?
- What exactly happens after you use Rufus? Does the drive disappear/become RAW in disk manager? Do you need to reboot for that to happen? Do you get an error?

kamikazilucas, 2023-09-02:
its visible in disk manager before
i just use it for windows i dont have any logs for windows
i need to reboot it for it to show up as raw
the drive just becomes corrupted im not in the drive rn and its showing this 
[image]

pbatard, 2023-09-02:
It's probably just the ESP (EFI System Partition) then.

Windows 10 and Windows 11 explicitly prevent access to the ESP (it's a Microsoft policy), and a Windows To Go needs to have an ESP (which you can see from the log as the first partition).

It may take a reboot for Windows to decide you should not be able to access it. And since this is system behaviour, it's not something Rufus can do much about, outside of changing the ESP to a general data partition, but then Windows To Go may not boot.