pbatard/rufus #2320: Feature Request: Multi Editions Provider

BNLOS, 2023-09-07:
Hey sorry for missing details but i hope you understand me i would like you to add a feature to enable multi editions for Windows ISOS you know
by adding ei.cfg file with the following contents
``
[Channel]
Retail
``

inside sources folder after flashing image to disk.

pineapple63, 2023-09-07:
I think this has been requested before, and there are no plans for this feature to be added (since such a feature would only benefit power users)

pbatard, 2023-09-08:
Yeah, `ei.cfg` is a power-user/admin feature, and as clearly indicated in the FAQ, I don't develop Rufus for power-users or admins: https://github.com/pbatard/rufus/wiki/FAQ#user-content-Why_cant_you_just_add_the_feature_I_want.

Besides, it appears to only be required in a __the very limited scenario__ where someone purchased a computer that uses a specific license of Windows, and purchased a second license later on, and Rufus is not designed to deal with licensing matters. Furthermore, when all you have to do is drop an `ei.cfg` file, and any code trying to inject a `ei.cfg` would have to take care of possible conflicts, and I also seem to remember that the mere presence of an `ei.cfg` file may result in the Windows installer behaving differently, I really see no incentive to add that feature in Rufus.

&rarr; Rejected

closed 2023-09-08 by pbatard