pbatard/rufus #2321: rufus Unable to recognize modern SSD USB drives

LoopholesInTheHeart, 2023-09-09:
I purchased a large capacity USB flash drive online, which is different from traditional USB flash drives and uses the same process as SSD.
The link to the product is https://item.jd.com/100040789478.html
On my computer, it is recognized as a disk drive and cannot be recognized by Rufus
As an example, on my computer, their attributes are
[image]
In Windows Device Manager and Disk Management, it looks like this

[image]

LoopholesInTheHeart, 2023-09-09:
Some Chinese translations in the window
可移动 -> removable, This is a removable mobile device
基本 -> basic, generic device, in this case, might mean unpluggable, like system disk0
in the right window
磁盘驱动器 -> Disk drive,Similar to a hard disk, it indicates the storage medium

pineapple63, 2023-09-09:
You will need to enable List USB Hard Drives, which is found under the advanced drive properties section

pbatard, 2023-09-09:
It's probably UASP, but as long as it's USB, Rufus should be able to access it without problem.

But since it's detected as an external large disk/HDD/SSD, Rufus will not list it by default, and you need to enable it as @pineapple63 mentioned. See https://github.com/pbatard/rufus/wiki/FAQ#user-content-Help_my_USB_drive_is_not_detected.

Considering that I am pretty sure this will answer your question, I will close this issue.

closed 2023-09-09 by pbatard