pbatard/rufus #2322: Rufus 4.2 fails to read a .VHD image created in Rufus 4.1

Simon311, 2023-09-10:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
A bootable VHD image created in Rufus 4.1 isn't recognized Rufus 4.2. Originally, the drive I've copied was also written by Rufus 4.1 and works flawlessly and the image in question is also recognized in Rufus 4.1 without issue. Logs from both versions below.

Log
---
Rufus x64 v4.2.2074 (Portable)
Windows version: Windows 10 Enterprise x64 (Build 19044.3324)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 2.0 device 'Teclast  CoolFlash USB Device' (FFFF:5678)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3824, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x2A589A36
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 29.3 GB (31456231424 bytes)
  Start Sector: 2048, Boot: No
Scanning image...
ISO analysis:
  'C:\Temp_\GlobalImage.vhd' doesn't look like an ISO image
Disk image analysis:
Could not open image 'C:\Temp_\GlobalImage.vhd': [0x00000570] The file or directory is corrupted and unreadable.
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3824, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x2A589A36
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 29.3 GB (31456231424 bytes)
  Start Sector: 2048, Boot: No
Failed to scan image

Rufus x64 v4.1.2045 (Portable)
Windows version: Windows 10 Enterprise x64 (Build 19044.3324)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Teclast  CoolFlash USB Device' (FFFF:5678)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3824, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x2A589A36
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 29.3 GB (31456231424 bytes)
  Start Sector: 2048, Boot: No
Scanning image...
ISO analysis:
  'C:\Temp_\GlobalImage.vhd' doesn't look like an ISO image
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a Fixed Hard Disk VHD file
  Image is a bootable VHD image
Using image: GlobalImage.vhd (55.9 GB)

Simon311, 2023-09-10:
It seems this was fixed in https://github.com/pbatard/rufus/commit/9c6b1ad97700cb834c7c4de96a84ed923758c034.
I suppose I could stick to 4.1 until 4.3 is out. Cheers.

closed 2023-09-10 by Simon311

pbatard, 2023-09-10:
Duplicate of #2309